chipsalliance/verible #1: Integrate with FuseSoC

mithro, 2019-11-12:
https://github.com/olofk/fusesoc
> FuseSoC is an award-winning package manager and a set of build tools for HDL (Hardware Description Language) code.
> 
> Its main purpose is to increase reuse of IP (Intellectual Property) cores and be an aid for creating, building and simulating SoC solutions.
> 
> FuseSoC makes it easier to
> * reuse existing cores
> * create compile-time or run-time configurations
> * run regression tests against multiple simulators
> * Port designs to new targets
> * let other projects use your code
> * set up continuous integration

Would be good to have FuseSoc able to use Verible as a linter and code formatter.

mithro, 2019-11-12:
@olofk

olofk, 2019-11-12:
Definitely. Adding a new Edalize backend for the linter and formatter should be straightforward

fangism, 2020-02-27:
FYI @msfschaffner @sjgitty
Already in use? done or needs more work?

imphil, 2020-02-27:
Verible lint is done in https://github.com/olofk/edalize/pull/100, verible format isn't done yet.