chipsalliance/verible #11: Set up Kokoro to build and run Verible tests

mithro, 2019-11-20:
I have a WIP pull request at https://github.com/google/verible/pull/9

mithro, 2019-11-20:
Currently the build fails with;
```
---------------------------------------------------------------
 Building Verible
---------------------------------------------------------------
+ bazel build --cxxopt=-std=c++17 //...
Starting local Bazel server and connecting to it...
INFO: Invocation ID: 732c2dec-e1c3-4576-ba5b-7934ec3ea320
Loading: 
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 18 packages loaded
    currently loading: bazel
Analyzing: 830 targets (19 packages loaded, 0 targets configured)
Analyzing: 830 targets (29 packages loaded, 271 targets configured)
Analyzing: 830 targets (36 packages loaded, 366 targets configured)
ERROR: /tmpfs/src/github/verible/bazel/BUILD:20:1: in configure_make rule //bazel:m4: 
Traceback (most recent call last):
	File "/tmpfs/src/github/verible/bazel/BUILD", line 20
		configure_make(name = 'm4')
	File "/tmpfs/tmp/bazel/external/rules_foreign_cc/tools/build_defs/configure.bzl", line 29, in _configure_make
		cc_external_rule_impl(ctx, attrs)
	File "/tmpfs/tmp/bazel/external/rules_foreign_cc/tools/build_defs/framework.bzl", line 209, in cc_external_rule_impl
		_define_out_cc_info(ctx, attrs, inputs, outputs)
	File "/tmpfs/tmp/bazel/external/rules_foreign_cc/tools/build_defs/framework.bzl", line 632, in _define_out_cc_info
		cc_common.create_compilation_context(headers = depset([outputs.out_in...]), <4 more arguments>)
parameter 'ctx' has no default value, in method call create_compilation_context(depset headers, depset system_includes, depset includes, depset quote_includes, depset defines) of 'cc_common'
ERROR: Analysis of target '//bazel:m4_bin' failed; build aborted: Analysis of target '//bazel:m4' failed; build aborted
INFO: Elapsed time: 10.000s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (47 packages loaded, 2581 targets configured)
FAILED: Build did NOT complete successfully (47 packages loaded, 2581 targets configured)
```

fangism, 2020-01-08:
Found another mention of this failure signature:

https://github.com/google/differential-privacy/issues/8

mithro, 2020-01-08:
I think we solved it at https://github.com/google/verible/pull/9 -- IE You need a newer bazel release (and make sure you are actually using it...)

closed 2020-01-08 by hzeller