chipsalliance/verible #15: Provide simple file-list flag

hzeller, 2019-11-20:
Right now, all  files are provided on the command-line, but it might be useful to have also the concept of a file-list. That way, it is possible to read e.g. an already preprocessed list of files to be further processed by verible (all tools).

The standardized way (common in verilog processing) looks like it is just to have a file that contains a list of command-line options, so not only a list of files, but everything that can be put on the command-line. Common flag name seems to be `-f` ?

Maybe there is already a simple way for the ABSL-flag library to read options from a file ?

fangism, 2019-11-21:
Behavior like: `tool $(cat flagfile) ...` ?

hzeller, 2019-11-22:
yes, essentially what the `*.f` files convention do in the EDA verilog world.