chipsalliance/verible #25: Interface and module port list formatting

msfschaffner, 2019-11-25:
This is how systemverilog interface and module declarations should look like:
```systemverilog
module my_mod (
  inout clk,
  inout rst_n
);

endmodule : my_mod

module my_intf (
  inout clk,
  inout rst_n
);

endmodule : my_intf

module my_mod #(
  parameter first_param = 1,
  parameter second_param = 2
) (
  inout clk,
  inout rst_n
);

endmodule : my_mod

module my_intf #(
  parameter first_param = 1,
  parameter second_param = 2
) (
  inout clk,
  inout rst_n
);

endmodule : my_intf
```

Verible currently formats them as follows:
```systemverilog

module my_mod(inout clk, inout rst_n);

endmodule : my_mod

module my_intf(inout clk, inout rst_n);

endmodule : my_intf

module my_mod #(parameter first_param = 1, parameter second_param = 2) (inout clk, inout rst_n);

endmodule : my_mod

module my_intf #(parameter first_param = 1, parameter second_param = 2) (inout clk, inout rst_n);

endmodule : my_intf

```

msfschaffner, 2019-11-25:
b/145151730

fangism, 2019-11-25:
Is there any misbehavior here that isn't covered by #23 and #24?

msfschaffner, 2019-11-25:
we usually do not write these declarations on one single line, so the line breaks should not be removed...

fangism, 2019-11-25:
Ah, if this is about avoiding single-line compaction, you can probably workaround this today by inserting a //-comment.

```
module foo (  //
    ...ports...
);
```

msfschaffner, 2019-11-25:
yes that works, but the question is whether we should make this behavior the default for ports and interfaces...

msfschaffner, 2019-11-25:
here's another example:
should:
```systemverilog
interface ibex_mem_intf #(
  parameter int ADDR_WIDTH = 32, 
  parameter int DATA_WIDTH = 32
);
```

is:
```systemverilog
interface ibex_mem_intf#(parameter int ADDR_WIDTH = 32, parameter int DATA_WIDTH = 32);
```

fangism, 2019-11-26:
Grammatically, both interface and module declarations are treated pretty much the same.

https://github.com/google/verible/blob/f74e900dbdd40c48fcaf1ff68db0c7f5bd342658/verilog/parser/verilog.y#L5061

fangism, 2019-11-26:
Note: currently missing partition boundary around module instance type actual parameters.

ldalek-antmicro, 2020-03-04:
It looks that that partition tree looks ok:
```
  { ([<auto>]) @{6}, policy: fit-else-expand
    { ([interface my_intf # (]) }
    { (>>>>[<auto>]) @{6,1}, policy: always-expand
      { (>>>>[parameter first_param = 1 ,]) }
      { (>>>>[parameter second_param = 2]) }
    }
    { ([) (]) }
    { (>>>>[<auto>]) @{6,3}, policy: always-expand
      { (>>>>[inout clk ,]) }
      { (>>>>[inout rst_n]) }
    }
    { ([) ;]) }
  }
  { ([endinterface : my_intf]) }
```
so to fix this issue enough would be to change the partition policy for
ports and parameters lists or am I missing something?

fangism, 2020-03-04:
> It looks that that partition tree looks ok:
> 
> ```
>   { ([<auto>]) @{6}, policy: fit-else-expand
>     { ([interface my_intf # (]) }
>     { (>>>>[<auto>]) @{6,1}, policy: always-expand
>       { (>>>>[parameter first_param = 1 ,]) }
>       { (>>>>[parameter second_param = 2]) }
>     }
>     { ([) (]) }
>     { (>>>>[<auto>]) @{6,3}, policy: always-expand
>       { (>>>>[inout clk ,]) }
>       { (>>>>[inout rst_n]) }
>     }
>     { ([) ;]) }
>   }
>   { ([endinterface : my_intf]) }
> ```
> 
> so to fix this issue enough would be to change the partition policy for
> ports and parameters lists [73f66fd](https://github.com/google/verible/commit/73f66fd48f745d2ef62fc1697d360924c3abe626) or am I missing something?

That looks right to me.

ldalek-antmicro, 2020-03-04:
Also I'm not sure about single ports and parameters like:
```
module my_mod (
    inout clk
);
```

I'm guessing they should be on its own line too?

And how about port connections? e.g.
```
module foo (
    .x(x),
    .y(y)
);
```

fangism, 2020-03-04:
> Also I'm not sure about single ports and parameters like:
> 
> ```
> module my_mod (
>     inout clk
> );
> ```
> 
> I'm guessing they should be on its own line too?
> 
> And how about port connections? e.g.
> 
> ```
> module foo (
>     .x(x),
>     .y(y)
> );
> ```

Yes.  Users have expressed that they would like consistent expansion of one per line, even when it fits.  This way port changes show up as whole-line diffs.

ldalek-antmicro, 2020-03-04:
Okay. Thanks for quick response!

fangism, 2020-03-04:
> Okay. Thanks for quick response!

Thanks for all the work on closely related issues!  This batch closes out most of the remaining issues our first small set of users are waiting on.

commit 55adfae3 (2020-03-04): PR #224: Fix Interface and module port list formatting
Fixes #25 #218

GitHub PR https://github.com/google/verible/pull/224

Copybara import of the project:

  - 73f66fd48f745d2ef62fc1697d360924c3abe626 Expand port and parameter lists in module and interface d... by Lukasz Dalek <ldalek@antmicro.com>
  - 552732b2b2c5314760233b6b473ef49c7e9641ff Put kPort in its own partition by Lukasz Dalek <ldalek@antmicro.com>
  - 0f1cbde322208f675c87f8a05b8762ecce58e354 Update module and interface declarations tests by Lukasz Dalek <ldalek@antmicro.com>

Closes #224

PiperOrigin-RevId: 298909899
closed 2020-03-04 by hzeller