chipsalliance/verible #26: Else statements need to start own partition

msfschaffner, 2019-11-25:
should:
```systemverilog
      if (trans_collected.read_write == WRITE)  item_collected_port.write(trans_collected); 
      else                                      collect_data_queue.put(trans_collected);
```
is:
```systemverilog
      if (trans_collected.read_write == WRITE)
        item_collected_port.write(trans_collected); else collect_data_queue.put(trans_collected);
```

msfschaffner, 2019-11-25:
b/144605105

ldalek-antmicro, 2019-12-03:
It looks like this commit https://github.com/google/verible/commit/36865ed868da4f141a2a8f8992eebc22824335f5 fixes the issue. @fangism can you confirm that?

fangism, 2019-12-03:
Doesn't look fixed yet.  Here's a reduced test case:

```systemverilog
task t; if (r == t) a.b(c); else d.e(f); endtask
```

I still get this output:

```systemverilog
task t;
  if (r == t) a.b(c); else d.e(f);
endtask
```

but we want:

```systemverilog
task t;
  if (r == t) a.b(c);
  else d.e(f);
endtask
```

fangism, 2019-12-04:
`verilog_format --show_token_partition_tree` actually looks correct:

```
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([task t ;]) }
  }
  { (  [<auto>]) @{1}
    { (  [<auto>]) @{1,0}
      { (  [if ( r == t )]) }
      { (    [<auto>]) @{1,0,1}
        { (    [a . b ( c ) ;]) }
      }
      { (  [else]) }
      { (    [<auto>]) @{1,0,3}
        { (    [d . e ( f ) ;]) }
      }
    }
  }
  { ([endtask]) }
}
Halting for diagnostic operation.
```

But what I think is happening in
https://github.com/google/verible/issues/26#issuecomment-561328376
is that it is choosing to fit on one line if possible.
We have to update the heuristic so that `else` forces the partition to be evaluated in pieces, rather than as a whole.

Here is the if-it-fits policy:

https://github.com/google/verible/blob/d34420461f422d1cdbf4585a376b5a3210b9f828/verilog/formatting/tree_unwrapper.cc#L624

Fit determination is here:

https://github.com/google/verible/blob/d34420461f422d1cdbf4585a376b5a3210b9f828/common/formatting/line_wrap_searcher.cc#L180

Note that `SpacingOptions::MustWrap` forces it to "not-fit".

Spacing policy (where static) is determined here:

https://github.com/google/verible/blob/d34420461f422d1cdbf4585a376b5a3210b9f828/verilog/formatting/token_annotator.cc#L451

Somewhere in this function, we should handle the `else` token case, but we currently don't.

Do you still want to take this issue, or shall I?

ldalek-antmicro, 2019-12-04:
Thanks for your response.
It looks like you pointed it out: token tree looks fine but it tries to fit that into single line.

How about handling `else` statement here:
https://github.com/antmicro/verible/commit/b38b1a58525d6adcf81831e3ef07a71318ea3190
it forces break in every condition except `begin end else ...` block?

> Do you still want to take this issue, or shall I?

If you want to try, please be my guest :smile:

fangism, 2019-12-04:
At first glance, I think https://github.com/antmicro/verible/commit/b38b1a58525d6adcf81831e3ef07a71318ea3190 will work.  I'll leave this assigned to you to confirm with testing.  :)

closed 2019-12-06 by hzeller