chipsalliance/verible #27: Indenting uvm macros with begin/end semantic

msfschaffner, 2019-11-25:
should:
```systemverilog
`uvm_object_utils_begin(ibex_mem_intf_seq_item)
  `uvm_field_int(addr, UVM_DEFAULT)
  `uvm_field_enum(rw_e, read_write, UVM_DEFAULT)
  `uvm_field_int(be, UVM_DEFAULT)
  `uvm_field_int(data, UVM_DEFAULT)
  `uvm_field_int(gnt_delay, UVM_DEFAULT)
  `uvm_field_int(rvalid_delay, UVM_DEFAULT)
  `uvm_field_int(error, UVM_DEFAULT)
`uvm_object_utils_end `uvm_object_new
```

is:
```systemverilog
`uvm_object_utils_begin(ibex_mem_intf_seq_item)
`uvm_field_int(addr, UVM_DEFAULT)
`uvm_field_enum(rw_e, read_write, UVM_DEFAULT)
`uvm_field_int(be, UVM_DEFAULT)
`uvm_field_int(data, UVM_DEFAULT)
`uvm_field_int(gnt_delay, UVM_DEFAULT)
`uvm_field_int(rvalid_delay, UVM_DEFAULT)
`uvm_field_int(error, UVM_DEFAULT)
`uvm_object_utils_end `uvm_object_new
```

msfschaffner, 2019-11-25:
b/145155336

ldalek-antmicro, 2019-12-20:
Hey @fangism,
I've fixed this issue by detecting `uvm_object_utils_begin` and `uvm_object_utils_end` leaves and by
indenting each 'kMacroCall' but I'm not sure if this approach is correct. I was thinking about changing
CST tree and indenting it like it's done between 'begin' and 'end' keywords.
Can you give me some hints how this should be done?

fangism, 2019-12-21:
There is certainly more than one way this could be accomplished.  And here is but one idea:

Allow macro calls to be processed and indented as-is, but in the fix-up phase after a visiting a node of type `kMacroCall` that matches `uvm.*end`, iterate backwards over the siblings until reaching `uvm.*begin` and adjust the indentation of partitions in between them.

mock up:
```
// assuming we've identified a begin/end pair that marks the range of interest...
for (auto& partition : /* range between begin/end pair */ ) {
 partition.ApplyPreOrder([&](UnwrappedLine& line) {
    line.SetIndentationSpaces(line.IndentationSpaces() + style_.indentation_spaces);
  });
}
```

references:
https://github.com/google/verible/blob/b1628f95aa58406a1ffa3f1d91354eae98d50930/common/util/vector_tree.h#L491
https://github.com/google/verible/blob/b1628f95aa58406a1ffa3f1d91354eae98d50930/common/formatting/unwrapped_line.h#L82

This *feels* simpler to me because 
1) we don't need to track any state during forward analysis (from `uvm.*begin`) (I'm not opposed to state-tracking.)
2) iterating through token partition siblings guarantees we find corresponding begin/end pairs (unless user's code is mismatched, in which case, we could just take no fix-up action)
3) it doesn't alter the shape of any existing partitions, only indentation amounts

The test cases for this should be rather interesting, there are many paired UVM macros like this.  We'll want to cover potentially weird cases like:

```
  `uvm_blah1_begin(...)
    ...  // indented more
  `uvm_blah1_end
  ...  // normal indentation
  `uvm_blah2_begin(...)
    ...  // indented more
  `uvm_blah2_end
  ...  // normal indentation
```

ldalek-antmicro, 2019-12-31:
Okay. First of all thanks for comment. It was very useful.
I've prepared PR #113 with changes.

I've decided to put every `uvm` macro in its own unwrapped line:
https://github.com/google/verible/pull/113/files#diff-7872d97b05fbc6f2a64cf14b8ae74d75R665
https://github.com/google/verible/pull/113/files#diff-7872d97b05fbc6f2a64cf14b8ae74d75R910

And then indent macros between `begin` and `end` keywords like you suggested:
https://github.com/google/verible/pull/113/files#diff-7872d97b05fbc6f2a64cf14b8ae74d75R924

commit 8a5b3130 (2020-01-03): PR #113: Indent macros inside `uvm_object_utils_begin-end
Fix #27

GitHub PR https://github.com/google/verible/pull/113

Copybara import of the project:

  - 1c2384eb9f53fff24768671f8de90c45114fe2f9 Indent `uvm_.*_begin-end macros by Lukasz Dalek <ldalek@antmicro.com>
  - e72d05f352121a16b617d28b060bfd0c9f77ef39 Add test cases for uvm macros indenting by Lukasz Dalek <ldalek@antmicro.com>

Closes #113

PiperOrigin-RevId: 287913196
closed 2020-01-03 by hzeller