chipsalliance/verible #29: [line breaks, wrap] Task arguments put on a new line

msfschaffner, 2019-11-25:
The issue is currently onliy occuring when a task/function has more than 1 arg.

should:
```systemverilog
  virtual task check_next_core_status(core_status_t core_status, 
                                      string error_msg = "",
                                      int timeout = 9999999);
```
is:
```systemverilog
  virtual task check_next_core_status(
      core_status_t core_status, string error_msg = "", int timeout = 9999999
  );
```

msfschaffner, 2019-11-25:
b/145157892

fangism, 2019-11-26:
Specifically, it aggressively wraps when it doesn't fit on a single line, at the moment.  It's not even using the wrap-optimizer at this time.  This is controlled by:
https://github.com/google/verible/blob/f74e900dbdd40c48fcaf1ff68db0c7f5bd342658/verilog/formatting/tree_unwrapper.cc#L660

ldalek-antmicro, 2019-12-16:
Hey @fangism,
Can you look at #90?

I've fixed this issue by skipping partitioning arguments in `tree_unwrapper.cc` and by
forcing line break after comma in kTaskHeader and kFunctionHeader. But unfortunately this approach
breaks some tests e.g. https://github.com/google/verible/pull/90/commits/6d7bfc63b979836dbbffde582ac085a377a919d6#diff-c44554418c4241ad93a4b74981f78b08L1350

Also I'm not sure about this test: :https://github.com/google/verible/pull/90/commits/6d7bfc63b979836dbbffde582ac085a377a919d6#diff-9886cf9de4b1ff6379851d2f4d707190L982

fangism, 2019-12-18:
Before we embark on any solution, we need to reach a consensus on what is acceptable.

@msfschaffner @snsokolov which of the following results would you consider acceptable?

1a) (from original comment) keep first port on same line (if it fits), align all other ports, close with last port
```systemverilog
  virtual task check_next_core_status(core_status_t core_status, 
                                      string error_msg = "",
                                      int timeout = 9999999);
```

1b) same as 1a, but closing on its own line
```systemverilog
  virtual task check_next_core_status(core_status_t core_status, 
                                      string error_msg = "",
                                      int timeout = 9999999
  );
```

1c) same as 1b, but closing on its own line, aligned to open-(
```systemverilog
  virtual task check_next_core_status(core_status_t core_status, 
                                      string error_msg = "",
                                      int timeout = 9999999
                                      );
```

2a) wrap first port, append the rest while they fit (allow more than one per line), close with last port
```systemverilog
  virtual task check_next_core_status(
      core_status_t core_status, string error_msg = "", int timeout = 9999999);
```

2b) same as 2a, but illustrated with ports wrapping, not enforcing one-per-line
```systemverilog
  virtual task check_next_core_status(
      core_status_t core_status, string error_msg = "", 
      int timeout = 9999999);
```

3a) wrap first port, force one port per line, close with last item
```systemverilog
  virtual task check_next_core_status(
    core_status_t core_status, 
    string error_msg = "",
    int timeout = 9999999);
```

3b) same as 3a, but close on own line
```systemverilog
  virtual task check_next_core_status(
    core_status_t core_status, 
    string error_msg = "",
    int timeout = 9999999
  );
```

ldalek-antmicro, 2020-01-22:
Hey @fangism,
I've created PR which does formatting like it's described in issue description. It's messy but it works (PR #148). e.g.
Input:
<pre>
class c; virtual task check_next_core_status(core_status_t core_status, string error_msg = "", int timeout = 9999999); endtask; endclass;
</pre>

Token tree:
<pre>
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([class c ;]) }
  }
  { (>>[<auto>]) @{1}
    { (>>[<auto>]) @{1,0}
      { (>>[virtual task check_next_core_status ( core_status_t core_status ,]) }
      { (>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>[string error_msg = "" ,]) }
      { (>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>[int timeout = 9999999 ) ;]) }
    }
    { (>>[endtask ;]) }
  }
  { ([endclass ;]) }
}
</pre>

Output:
<pre>
class c;
  virtual task check_next_core_status(core_status_t core_status,
                                      string error_msg = "",
                                      int timeout = 9999999);
  endtask;
endclass;
</pre>

At this moment we can proceed with it or we can drop it and proceed with `3b` (which is easy to do).
What do you think?

fangism, 2020-01-23:
@ldalek-antmicro Thanks for taking a stab at this.
The token-partition phase was never intended to make detailed wrapping/indentation decisions, but rather convey the notion of a hierarchy of opportunities and strategies.  I know it may seem vague where the line is drawn because of the existence of `kFitOnLineElseExpand` as a quick heuristic.

https://github.com/google/verible/blob/18c537948ac92e05010f02f0fbe8f3e2028f07d5/common/formatting/unwrapped_line.h#L36

Today, the finalized partitions (part determined by `kFitOnLineElseExpand`) are all fed into the same line-wrap-search optimization as a flat worklist, but my intent was to extend that to more strategies, including coordinated alignment across sub-partitions, and even treating whole partitions as units for determining wrapping which is most relevant to this issue.

Using the above example, `[core_status_t core_status ,]` would be considered a unit, as would `[int timeout = 999999 ) ;]`.  Each unit's size would be considered their most compact length, by formatting rules.  A very simple heuristic at the partition that considered the whole task header would only have to make a binary decision after `(`: wrap the first unit, or continue it on the same line, which looks like:

```systemverilog
class c;
  virtual task check_next_core_status(core_status_t core_status,
                                      string error_msg = "",
                                      int timeout = 9999999);
  endtask
endclass
```

or:

```systemverilog
class c;
  virtual task check_next_core_status(
      core_status_t core_status, string error_msg = "", int timeout = 9999999);
  endtask
endclass
```

Basically, all units after the first, are simply appended while they fit, wrapped when they don't (greedy, linear).
This binary decision at the first unit is much cheaper than doing a token-by-token search over the space of tokens spanning all of the units (here parameter declarations).  One could choose whichever results the fewest lines (minimum vertical space), for example.

Long story short, I would like to introduce new context-dependent formatting strategies where decisions can be simplified by using coarser-grain units, and a full search/optimization is not warranted.

Does this help?

ldalek-antmicro, 2020-01-24:
Today formatting is done in three steps:
1. Building partition token tree:
https://github.com/google/verible/blob/master/verilog/formatting/formatter.cc#L244

2. Flatting tree structure to list of unwrapped lines:
https://github.com/google/verible/blob/master/verilog/formatting/formatter.cc#L266
by using `ExpandableTreeView` 

3. Wrapping unwrapped lines created in previous step https://github.com/google/verible/blob/master/verilog/formatting/formatter.cc#L278
based on token annotation created before 1st step https://github.com/google/verible/blob/master/verilog/formatting/formatter.cc#L232

So for proper indentation we should introduce new partition policy, for example
`kColumnarAlignment` https://github.com/google/verible/blob/master/common/formatting/unwrapped_line.h#L42.

Keep current token tree partitioning:
<pre>
    { (>>[<auto>]) @{1,0}
      { (>>[virtual task check_next_core_status (]) }
      { (>>>>>>[<auto>]) @{1,0,1}
        { (>>>>>>[core_status_t core_status ,]) }
        { (>>>>>>[string error_msg = "" ,]) }
        { (>>>>>>[int timeout = 9999999]) }
      }
      { (>>[) ;]) }
    }
</pre>

Treat `kColumnarAlignment` as `kAlwaysExpand` and keep `ExpandableTreeView` untouched so we get such output:
<pre>
uw: [class c ;]
uw: >>[virtual task check_next_core_status (]
uw: >>>>>>>>[core_status_t core_status ,]
uw: >>>>>>>>[string error_msg = "" ,]
uw: >>>>>>>>[int timeout = 9999999]
uw: >>[) ;]
uw: >>[endtask ;]
uw: [endclass ;]
</pre>

and modify this loop https://github.com/google/verible/blob/master/verilog/formatting/formatter.cc#L274 so it would support `kColumnarAlignment` policy and format unwrapped lines like this:
<pre>
  virtual task check_next_core_status(core_status_t core_status,
                                      string error_msg = "",
                                      int timeout = 9999999);
</pre>

To sum up:
1. Leave `tree_uwrapper()` as it is (except adding `kColumnarAlignment`)
2. Leave `ExpandableTreeView` as it is (except adding `kColumnarAlignment`
3. Modify loop so it properly indent lines marked by `kColumnarAlignment` policy

Did I understand correctly?

fangism, 2020-01-25:
Your understanding of the formatter phases is correct.  (And apologies for not having written up the documentation for this yet.)

The partition tree phase is unlikely to change, apart from handling more cases.
Flattening a markup of the partition tree is expected to evolve.
Wrapping will continue to operate on the partitions that need it, but other strategies will be implemented.

I wouldn't call the new strategy for this issue `ColumnarAlignment`, though -- let's reserve that for when we get around to actual table-looking alignment across partitions.
This has a greedy characteristic (appending while fitting), and operates on whole partitions (instead of tokens) as a unit.  `kAppendFittingSubpartitions`?  Naming is hard.

If we decided to force one partition per line, that would be `kAlwaysExpand` at a particular level.  However, I think we're not confined to one parameter per line for function/task headers.

ldalek-antmicro, 2020-01-27:
Okay, new approach. I've added new partition policy: `kAppendFittingSubpartitions` (for now, might change later). I'm applying this policy to `kTaskHeader` node & subnodes https://github.com/google/verible/pull/148/files#diff-7872d97b05fbc6f2a64cf14b8ae74d75R946. Then all partitions with `kAppendFittingSubpartitions` policy are expanded https://github.com/google/verible/pull/148/files#diff-7872d97b05fbc6f2a64cf14b8ae74d75R946 (like in `kAlwaysExpand`) and then lines are properly indented https://github.com/google/verible/pull/148/files#diff-7872d97b05fbc6f2a64cf14b8ae74d75R946.

So it looks right now like this:
Input:
<pre>
class c; virtual task check_next_core_status(core_status_t core_status, string error_msg = "", int timeout = 9999999); endtask; endclass;
class d; virtual task foo(aa bb, string cc = "", int dd = 9); endtask; endclass;
</pre>

Token partition tree:
<pre>
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([class c ;]) }
  }
  { (>>[<auto>]) @{1}
    { (>>[<auto>]) @{1,0}
      { (>>[virtual task check_next_core_status (]) }
      { (>>>>>>[<auto>]) @{1,0,1}
        { (>>>>>>[core_status_t core_status ,]) }
        { (>>>>>>[string error_msg = "" ,]) }
        { (>>>>>>[int timeout = 9999999]) }
      }
      { (>>[) ;]) }
    }
    { (>>[endtask ;]) }
  }
  { ([endclass ;]) }
  { ([<auto>]) @{3}
    { ([class d ;]) }
  }
  { (>>[<auto>]) @{4}
    { (>>[<auto>]) @{4,0}
      { (>>[virtual task foo (]) }
      { (>>>>>>[<auto>]) @{4,0,1}
        { (>>>>>>[aa bb ,]) }
        { (>>>>>>[string cc = "" ,]) }
        { (>>>>>>[int dd = 9]) }
      }
      { (>>[) ;]) }
    }
    { (>>[endtask ;]) }
  }
  { ([endclass ;]) }
}
Halting for diagnostic operation.
</pre>

Partitions policy:
<pre>
Token partition policy:
 :: [class c ;] :: fit-else-expand
 :: >>[virtual task check_next_core_status (] :: append-fitting-subpartitions
 :: >>>>>>[core_status_t core_status ,] :: append-fitting-subpartitions
 :: >>>>>>[string error_msg = "" ,] :: append-fitting-subpartitions
 :: >>>>>>[int timeout = 9999999] :: append-fitting-subpartitions
 :: >>[) ;] :: append-fitting-subpartitions
 :: >>[endtask ;] :: always-expand
 :: [endclass ;] :: always-expand
 :: [class d ;] :: fit-else-expand
 :: >>[virtual task foo (] :: append-fitting-subpartitions
 :: >>>>>>[aa bb ,] :: append-fitting-subpartitions
 :: >>>>>>[string cc = "" ,] :: append-fitting-subpartitions
 :: >>>>>>[int dd = 9] :: append-fitting-subpartitions
 :: >>[) ;] :: append-fitting-subpartitions
 :: >>[endtask ;] :: always-expand
 :: [endclass ;] :: always-expand
Halting for diagnostic operation.
</pre>

Output:
<pre>
class c;
  virtual task check_next_core_status(core_status_t core_status,
                                      string error_msg = "",
                                      int timeout = 9999999);
  endtask;
endclass;
class d;
  virtual task foo(aa bb,
                   string cc = "",
                   int dd = 9);
  endtask;
endclass;
</pre>

Right now I'm thinking about fixing `kAppendFittingSubpartitions` policy.
I'd like to expand partitions only when parameters don't fit.

What do you think @fangism ?

fangism, 2020-01-27:
So far, the output looks acceptable (I understand that this is without compaction).

As far as partitioning the task header, I would also consider merging `);` into the previous leaf partition, to treat `last port);` as a single unit.  (Just like we have trailing commas attached to earlier port declarations' partitions.)

I'll look through the draft PR for more details.

fangism, 2020-02-04:
https://github.com/google/verible/pull/148#issuecomment-581972774
captures the updated plan-of-record around this issue.
Summary: will try to greedily compact subpartitions (here, ports) into same line while it fits, wrapping only when necessary.   There will be a single binary decision whether to wrap the first element, and others will just follow.

commit 0fb112c1 (2020-03-05): PR #148: Put task arguments on a new line
Fixes #29

GitHub PR https://github.com/google/verible/pull/148

Copybara import of the project:

  - ce43c85dc8a740f334cd5a85cf0e619e80337a68 Apply kAppendFittingSupartitions policy to task and funct... by Lukasz Dalek <ldalek@antmicro.com>
  - 7a53f0eff662cb87fe42f5a13892d1b509d9a78b Apply kAppendFittingSupartitions policy to class construc... by Lukasz Dalek <ldalek@antmicro.com>

Closes #148

PiperOrigin-RevId: 298989116
closed 2020-03-05 by hzeller