chipsalliance/verible #30: [wrap penalty] Awkward $sformatf() line break

msfschaffner, 2019-11-25:
should:
```systemverilog
      trans_collected.be   = vif.be;
      `uvm_info(get_full_name(), $sformatf("Detect request with address: %0x",
                trans_collected.addr), UVM_HIGH)
```
is:
```systemverilog
      trans_collected.be = vif.be;
      `uvm_info(get_full_name(), $sformatf("Detect request with address: %0x", trans_collected.addr
                ), UVM_HIGH)

```

msfschaffner, 2019-11-25:
b/145157871

msfschaffner, 2019-11-26:
Here is another example that requires some tuning:

should:
```systemverilog
case (ref_type)
  default: begin
    `uvm_fatal(`gfn, $sformatf("The data 0x%0h written to the signature address is formatted 
                               incorrectly.", signature_data))
  end
endcase
```
this would also be permissible:
```systemverilog
case (ref_type)
  default: begin
    `uvm_fatal(`gfn, $sformatf("The data 0x%0h written to the signature address is formatted 
          incorrectly.", signature_data))
  end
endcase
```
is:
```systemverilog
case (ref_type)
  default: begin
    `uvm_fatal(`gfn, $sformatf(
               "The data 0x%0h written to the signature address is formatted incorrectly.",
                   signature_data))
  end
endcase 
```

fangism, 2019-11-26:
Acknowledged.  There has been zero-effort to-date in choosing rational penalty values.

ldalek-antmicro, 2021-06-08:
Extra example:
[hw/ip/aes/dv/env/seq_lib/aes_stress_vseq.sv](https://github.com/lowRISC/opentitan/blob/master/hw/ip/aes/dv/env/seq_lib/aes_stress_vseq.sv#L13)
```diff
-    `uvm_info(`gfn, $sformatf("\n\n\t ----| STARTING AES MAIN SEQUENCE |----\n %s",
-                              cfg.convert2string()), UVM_LOW)
+    `uvm_info(`gfn, $sformatf(
+              "\n\n\t ----| STARTING AES MAIN SEQUENCE |----\n %s", cfg.convert2string()), UVM_LOW)

```

ldalek-antmicro, 2021-07-20:
Example results of `./utils/verible-format.py -l` (narrowed to this PR):
https://github.com/antmicro/opentitan/commit/44707dcf7a40a215ba00b82db3b6be66211bbb81

PR #1045 (2021-11-04): Use Layout Optimizer to format nested function calls
Depends on: https://github.com/chipsalliance/verible/pull/1044
Fixes #30 

## Example

Input:

```systemverilog
`uvm_info(get_full_name(), $sformatf("Detect request with address: %0x", trans_collected.addr), UVM_HIGH)

`uvm_fatal(`gfn, $sformatf("The data 0x%0h written to the signature address is formatted incorrectly.", signature_data))
```

Results:

indent: 0 spaces
```systemverilog
`uvm_info(get_full_name(), $sformatf("Detect request with address: %0x", trans_collected.addr),
          UVM_HIGH)
`uvm_fatal(
    `gfn, $sformatf(
    "The  [... cut]
closed 2021-11-04 by hzeller