chipsalliance/verible #32: [Indentation] Multiline list parameter

msfschaffner, 2019-11-25:
should:
```systemverilog
// Implemented previlieged CSR list
parameter privileged_reg_t implemented_csr[] = {
  // Machine mode mode CSR
  MVENDORID,        // Vendor ID
  MARCHID,          // Architecture ID
  DSCRATCH0,        // Debug scratch register 0
  DSCRATCH1         // Debug scratch register 1
};
```
is:
```systemverilog
// Implemented previlieged CSR list
parameter privileged_reg_t implemented_csr[] = {
// Machine mode mode CSR
MVENDORID,  // Vendor ID
MARCHID,  // Architecture ID
DSCRATCH0,  // Debug scratch register 0
DSCRATCH1  // Debug scratch register 1
};
```

msfschaffner, 2019-11-25:
b/145160278

fangism, 2019-11-26:
Confirmed.  Need to consult style guide on whether braced-lists should have their items indented +2 or +4 spaces.

fangism, 2020-02-27:
The RHS is a concatenation expression.  From #219 this is not being partitioned properly, so that should be handled first.  The new partitions should be indented 2 spaces (for now).

commit fd1cbad5 (2020-07-20): PR #355: Expand concatenation expressions' partitioning
Fixes #219, fixes #32.

GitHub PR https://github.com/google/verible/pull/355

Copybara import of the project:

  - aa69977baf38e6579dee6e267e8db9ec84215da2 Create new partitions for concatenation items by Rafal Kapuscik <rkapuscik@antmicro.com>
  - b1d2f491a8c4324e5bea2231bc8e8b9f1a18cc51 Attach trailing commas, add more tests by Rafal Kapuscik <rkapuscik@antmicro.com>
  - 23b99e841686d3d47ec018c39787a743a665749d Add ConcatenationExpression to grammar by Rafal Kapuscik <rkapuscik@antmicro.com>
  - 8981f15ed8f6445535ba198a673f7922abe0c256  [... cut]
closed 2020-07-20 by hzeller