chipsalliance/verible #33: [line breaks, wraps] Multiline parameter break tuning

msfschaffner, 2019-11-25:
should:
```systemverilog
parameter exception_cause_t implemented_exception[] = {INSTRUCTION_ACCESS_FAULT, 
                                                       ILLEGAL_INSTRUCTION, BREAKPOINT,
                                                       LOAD_ACCESS_FAULT, STORE_AMO_ACCESS_FAULT, 
                                                       ECALL_MMODE};

```
is:
```systemverilog
parameter exception_cause_t
    implemented_exception[] = {INSTRUCTION_ACCESS_FAULT, ILLEGAL_INSTRUCTION, BREAKPOINT,
                               LOAD_ACCESS_FAULT, STORE_AMO_ACCESS_FAULT, ECALL_MMODE};
```

Note that this "compact" enumeration style is different than from the "long list style" mentioned in #32

fangism, 2019-11-26:
This look like a case that deferred to the wrap-optimizer, which is currently subject to zero-effort penalty tuning.  We should be able to achieve the form you desire with better penalty values.  Will categorize as such.

How amenable would you be to:

```
parameter foo_t bar[] = {
  value,
  value,
  ...
};
```

or

```
parameter foo_t bar[] = {
  value, value,
  ...
};
```

msfschaffner, 2019-11-26:
So your first example would basically correspond to the "long list style" in #32, right? I am actually fine with both styles that you suggested. is it possible to support both of them? I.e. variant 1) for long lists and 2) for short ones? or should we distinguish the two cases by adding inline `//` blockers if we desire style 1), and default to 2) otherwise?

msfschaffner, 2019-11-26:
b/145161192