chipsalliance/verible #34: Parameterized module instantiation

msfschaffner, 2019-11-26:
```systemverilog
  // CSR access interface
  core_ibex_csr_if csr_if(.clk(clk));

  ibex_core_tracing #(.DmHaltAddr(`BOOT_ADDR + 'h0), .DmExceptionAddr(`BOOT_ADDR + 'h4))
      dut(
          .clk_i(clk),
          .rst_ni(rst_n),
          .test_en_i(1'b1),
          .hart_id_i(32'b0),
          .boot_addr_i(`BOOT_ADDR),  // align with spike boot address
          .irq_software_i(irq_vif.irq_software),
          .irq_timer_i(irq_vif.irq_timer),
          .irq_external_i(irq_vif.irq_external),
          .irq_fast_i(irq_vif.irq_fast),
          .irq_nm_i(irq_vif.irq_nm),
          .fetch_enable_i(dut_if.fetch_enable),
          .debug_req_i(dut_if.debug_req),
          .data_gnt_i(data_mem_vif.grant),
          .data_rvalid_i(data_mem_vif.rvalid),
          .data_rdata_i(data_mem_vif.rdata),
          .data_err_i(data_mem_vif.error),
          .instr_gnt_i(instr_mem_vif.grant),
          .instr_rvalid_i(instr_mem_vif.rvalid),
          .instr_rdata_i(instr_mem_vif.rdata),
          .instr_err_i(instr_mem_vif.error)
      );
```

msfschaffner, 2019-11-26:
Related issues: #25 #24 #23

msfschaffner, 2019-11-26:
b/145160998

fangism, 2019-11-26:
This is currently partitioned the way it is because, in general, there may be multiple instances in a single instantiation item (sharing the same type).  In practice, we rarely see more than one instance at a time, so perhaps we could make a special-case out of this for singleton instances.  This would effectively flatten one level of partitioning so that the whole instantiation statement would be considered as a unit, and sub-divided to the form you desire if it doesn't just fit on a single line.

fangism, 2019-11-26:
Illustration:

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/formatting/b145160998.sv
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([module m ;]) }
  }
  { (  [<auto>]) @{1}
    { (  [<auto>]) @{1,0}
      { (  [ibex_core_tracing # ( . DmHaltAddr ( `BOOT_ADDR + 'h 0 ) , . DmExceptionAddr ( `BOOT_ADDR + 'h 4 ) )]) }
      { (      [<auto>]) @{1,0,1}         // THIS node could be flattened when it is a singleton
        { (      [dut (]) }
        { (          [<auto>]) @{1,0,1,1}
          { (          [. clk_i ( clk ) ,]) }
          { (          [. rst_ni ( rst_n ) ,]) }
          { (          [. test_en_i ( 1 'b 1 ) ,]) }
          { (          [. instr_rdata_i ( instr_mem_vif . rdata ) ,]) }
          { (          [. instr_err_i ( instr_mem_vif . error )]) }
        }
        { (      [) ;]) }
      }
    }
  }
  { ([endmodule]) }
}
```

Also, there is probably a partition boundary missing around the module type parameter actuals, related to #25.

msfschaffner, 2019-11-26:
Indeed, we almost never instantiate more than one single instance (when it comes to modules and interfaces). in fact, in terms of style I would even lean towards enforcing singletons for clarity.

fangism, 2019-12-12:
Blocked-by #86 #87

fangism, 2019-12-12:
Once #86 and #87 are fixed, we will revisit end-to-end examples, and aim to produce the formatting in the original comment, perhaps requiring a single //-comment for force partition expansion.

msfschaffner, 2019-12-16:
Another related example:

should:
```systemverilog
module gpiodpi #(
    parameter string NAME = "gpio0", 
    parameter N_GPIO = 32
) (
    input logic clk_i,
    input logic rst_ni,

    output logic [N_GPIO-1:0] gpio_p2d,
    input logic [N_GPIO-1:0] gpio_d2p,
    input logic [N_GPIO-1:0] gpio_en_d2p
);
```

is:
```systemverilog
module gpiodpi
#(
    parameter string NAME = "gpio0", parameter N_GPIO = 32
) (
    input logic clk_i,
    input logic rst_ni,

    output logic [N_GPIO-1:0] gpio_p2d,
    input logic [N_GPIO-1:0] gpio_d2p,
    input logic [N_GPIO-1:0] gpio_en_d2p
);
```

fangism, 2019-12-16:
Acknowledged as #86.

msfschaffner, 2019-12-16:
related, but should as far as I remember be formatted according to function/task style:

should:
```systemverilog
class cip_base_scoreboard #(type RAL_T = dv_base_reg_block,
                            type CFG_T = cip_base_env_cfg,
                            type COV_T = cip_base_env_cov)
```

is:
```systemverilog
class cip_base_scoreboard #(
    type RAL_T = dv_base_reg_block, type CFG_T = cip_base_env_cfg, type COV_T = cip_base_env_cov
)
```

fangism, 2019-12-16:
Context-sensitive formatting policies for formal parameter lists.  Fun!  I'd file that as a separate issue, specific to `class`es.  I'm not sure our internal style guide says anything about formatting class parameters.

commit 7eaae839 (2019-12-23): Partition single-instance declarations as a single unit (flatten; fuse-sibling).

Single instance declarations are now partitioned (examples):

  [foo bar();]

  [foo #() bar();]

  [foo #(]
  >>[...]
  [) bar ();]

  [foo #() bar (]
  >>[...]
  [);]

  [foo #(]
  >>[...]
  [) bar (]
  >>[...]
  [);]

For forcing expansion to one .named(port) per line, use //-comment.

fixes #87
fixes #34

PiperOrigin-RevId: 286910778
closed 2019-12-23 by hzeller