chipsalliance/verible #35: [macro] Macro argument alignment/placement

msfschaffner, 2019-11-26:
should:
```systemverilog
  virtual task tl_access_sub();
    `DV_SPINWAIT(
        // thread to read/write tlul
        tl_host_single_seq  tl_seq;
        `uvm_create_on(tl_seq, p_sequencer.tl_sequencer_h)
        if (cfg.zero_delays) begin
        tl_seq.min_req_delay = 0;
        tl_seq.max_req_delay = 0;
        end
        csr_utils_pkg::increment_outstanding_access();
        `DV_CHECK_RANDOMIZE_WITH_FATAL(tl_seq,
            addr      == local::addr;
            size      == local::size;
            mask      == local::mask;
              if (write) {
                if ($countones(mask) < (1 << size)) opcode == tlul_pkg::PutPartialData;
                else opcode inside {tlul_pkg::PutPartialData, tlul_pkg::PutFullData};
                data    == local::data;
                } else {
                  opcode  == tlul_pkg::Get;
                })
        `uvm_send(tl_seq)
        if (!write) data = tl_seq.rsp.d_data;
        if (check_rsp) begin
          `DV_CHECK_EQ(tl_seq.rsp.d_error, exp_err_rsp, "unexpected error response")
        end
        csr_utils_pkg::decrement_outstanding_access();,
        // thread to check timeout
        $sformatf("Timeout waiting tl_access : addr=0x%0h", addr))
  endtask
```

is:
```systemverilog
   virtual task tl_access_sub();
    `DV_SPINWAIT(
    // thread to read/write tlul
    tl_host_single_seq  tl_seq;
        `uvm_create_on(tl_seq, p_sequencer.tl_sequencer_h)
        if (cfg.zero_delays) begin
        tl_seq.min_req_delay = 0;
        tl_seq.max_req_delay = 0;
        end
        csr_utils_pkg::increment_outstanding_access();
        `DV_CHECK_RANDOMIZE_WITH_FATAL(tl_seq,
            addr      == local::addr;
            size      == local::size;
            mask      == local::mask;
              if (write) {
                if ($countones(mask) < (1 << size)) opcode == tlul_pkg::PutPartialData;
                else opcode inside {tlul_pkg::PutPartialData, tlul_pkg::PutFullData};
                data    == local::data;
                } else {
                  opcode  == tlul_pkg::Get;
                })
        `uvm_send(tl_seq)
        if (!write) data = tl_seq.rsp.d_data;
        if (check_rsp) begin
          `DV_CHECK_EQ(tl_seq.rsp.d_error, exp_err_rsp, "unexpected error response")
        end
        csr_utils_pkg::decrement_outstanding_access();
        ,
    // thread to check timeout
    $sformatf("Timeout waiting tl_access : addr=0x%0h", addr))
  endtask
```

msfschaffner, 2019-11-26:
b/145166415

fangism, 2019-11-26:
This looks odd.  I expect the lexer/parser to leave macro arguments un-lexed, and remain as one large atomic blob of text.

```
$ verilog_syntax --printtokens ~/testing/verilog/b145166415.sv
Lexed and filtered tokens: 
(#"class" @0-5: "class")
(#SymbolIdentifier @6-7: "c")
(#';' @7-8: ";")
(#"virtual" @11-18: "virtual")
(#"task" @19-23: "task")
(#SymbolIdentifier @24-37: "tl_access_sub")
(#'(' @37-38: "(")
(#')' @38-39: ")")
(#';' @39-40: ";")
(#MacroCallId @45-57: "`DV_SPINWAIT")
(#'(' @57-58: "(")
(#MacroArg @104-536: "tl_host_single_seq  tl_seq;
        `uvm_create_on(tl_seq, p_sequencer.tl_sequencer_h)
        if (cfg.zero_delays) begin
          tl_seq.min_req_delay = 0;
          tl_seq.max_req_delay = 0;
        end
        if (!write) data = tl_seq.rsp.d_data;
        if (check_rsp) begin
          `DV_CHECK_EQ(tl_seq.rsp.d_error, exp_err_rsp, "unexpected error response")
        end
        csr_utils_pkg::decrement_outstanding_access();")
(#',' @536-537: ",")
(#SystemTFIdentifier @581-590: "$sformatf")
(#'(' @590-591: "(")
(#TK_StringLiteral @591-631: ""Timeout waiting tl_access : addr=0x%0h"")
(#',' @631-632: ",")
(#SymbolIdentifier @633-637: "addr")
(#')' @637-638: ")")
(#MacroCallCloseToEndLine @638-639: ")")
(#"endtask" @642-649: "endtask")
(#"endclass" @650-658: "endclass")
(#$end @659-659: "")
```

Somehow, this line `tl_host_single_seq  tl_seq;` lost indentation.

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/b145166415.sv
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([class c ;]) }
  }
  { (  [<auto>]) @{1}
    { (  [<auto>]) @{1,0}
      { (  [virtual task tl_access_sub ( ) ;]) }
    }
    { (    [<auto>]) @{1,1}
      { (    [`DV_SPINWAIT (]) }
      { (    [// thread to read/write tlul]) }
      { (    [tl_host_single_seq  tl_seq;
        `uvm_create_on(tl_seq, p_sequencer.tl_sequencer_h)
        if (cfg.zero_delays) begin
          tl_seq.min_req_delay = 0;
          tl_seq.max_req_delay = 0;
        end
        if (!write) data = tl_seq.rsp.d_data;
        if (check_rsp) begin
          `DV_CHECK_EQ(tl_seq.rsp.d_error, exp_err_rsp, "unexpected error response")
        end
        csr_utils_pkg::decrement_outstanding_access(); ,]) }
      { (    [// thread to check timeout]) }
      { (    [$sformatf ( "Timeout waiting tl_access : addr=0x%0h" , addr ) )]) }
    }
    { (  [endtask]) }
  }
  { ([endclass]) }
}
```

It looks like the leading comment is still lexed as its own token, separated from the rest of the macro argument.  And then the spacing before `tl_host_single_seq  tl_seq;` is ignored.  

I think the safest quick way forward is to preserve spacing in the context of macro arguments, including around comments (even if they are lexed separately).  This should be feasible with context-sensitive inter-token annotation.

commit c53e878e (2020-01-08): PR #89: Format macro argument list
Fixes #35.

GitHub PR https://github.com/google/verible/pull/89

Copybara import of the project:

  - b5af58eeff1ebb4978337b4bfd348dfd2cc55ef1 Keep macro arguments` formatting by Rafal Kapuscik <rkapuscik@antmicro.com>
  - 5a3bedfe5449d3e3b0363609485cad216be49f23 Indent macro arguments when expanding by Rafal Kapuscik <rkapuscik@antmicro.com>
  - 5ea75b7466e19381e39be8f7b1d5cb9a1e8a5fd3 Update tree_unwrapper tests by Rafal Kapuscik <rkapuscik@antmicro.com>
  - 2669e84393b980364dd44a740e9f1775e353ae54 Add nested macro call end-to-end tests by Rafal Kapuscik <r [... cut]
closed 2020-01-08 by hzeller