chipsalliance/verible #39: [token partitioning] group else-if together

msfschaffner, 2019-11-26:
Filed on behalf of @fangism.

bad:
```systemverilog
function void foo();
    if ($value$plusargs("target_mabist_group=%s", target_mabist_group)) begin
      if (target_mabist_group == "bo") begin
        target_mabist = $urandom_range(kMabistBo0, kMabistBo10);
      end else
      if (target_mabist_group == "cc") begin
        target_mabist = $urandom_range(kMabistCc0, kMabistCc2);
      end else
      if (target_mabist_group == "cpu") begin
        target_mabist = kMabistCpu;
      end else    ...
```

good:
```systemverilog
function void foo();
    if ($value$plusargs("target_mabist_group=%s", target_mabist_group)) begin
      if (target_mabist_group == "bo") begin
        target_mabist = $urandom_range(kMabistBo0, kMabistBo10);
      end else if (target_mabist_group == "cc") begin
        target_mabist = $urandom_range(kMabistCc0, kMabistCc2);
      end else if (target_mabist_group == "cpu") begin
        target_mabist = kMabistCpu;
      end else begin    ...
```

msfschaffner, 2019-11-26:
b/143177175

rkapuscik, 2019-12-04:
I don't reproduce this behavior on d553b8a (master branch - newest d344204 does not build), the output is the same as in good example.
@msfschaffner could you confirm that this is fixed?

fangism, 2019-12-04:
I'm checking with this reduced test case:

```systemverilog
function void foo();
if (t == "b") begin x = y; end else if (t == "c") begin x = z; end
endfunction
```

which produces

```systemverilog
function void foo();
  if (t == "b") begin
    x = y;
  end else if (t == "c") begin
    x = z;
  end
endfunction
```

which looks good to me.

Further diagnostics `verilog_format --show_token_partition_tree FILE`:

```
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([function void foo ( ) ;]) }
  }
  { (  [<auto>]) @{1}
    { (  [<auto>]) @{1,0}
      { (  [if ( t == "b" ) begin]) }
      { (    [<auto>]) @{1,0,1}
        { (    [x = y ;]) }
      }
      { (  [end else if ( t == "c" ) begin]) }
      { (    [<auto>]) @{1,0,3}
        { (    [x = z ;]) }
      }
      { (  [end]) }
    }
  }
  { ([endfunction]) }
}
Halting for diagnostic operation.
```

The `end else if` partition looks correct.

I will review unit/integration tests to see if this is properly covered (add some if they are not) and close this.

closed 2019-12-04 by hzeller

msfschaffner, 2019-12-05:
Sorry to be late to the party. I ran this again with the newest build using 
```systemverilog
function void foo();
  if ($value$plusargs("target_mabist_group=%s", target_mabist_group)) begin
    if (target_mabist_group == "bo") begin
      target_mabist = $urandom_range(kMabistBo0, kMabistBo10);
    end else 
    if (target_mabist_group == "cc") begin
      target_mabist = $urandom_range(kMabistCc0, kMabistCc2);
    end else 
    if (target_mabist_group == "cpu") begin
      target_mabist = kMabistCpu;
    end
  end
endfunction
```
which produces 
```systemverilog
function void foo();
  if ($value$plusargs("target_mabist_group=%s", target_mabist_group)) begin
    if (target_mabist_group == "bo") begin
      target_mabist = $urandom_range(kMabistBo0, kMabistBo10);
    end else if (target_mabist_group == "cc") begin
      target_mabist = $urandom_range(kMabistCc0, kMabistCc2);
    end else if (target_mabist_group == "cpu") begin
      target_mabist = kMabistCpu;
    end
  end
endfunction
```
which LGTM.

fangism, 2019-12-05:
Thanks for confirming, Michael.

commit 4e5b485b (2019-12-04): Add test cases to cover else-if pairing.

fixes #39

PiperOrigin-RevId: 283833403