chipsalliance/verible #40: [indentation] 4 space vs 2 space indentation in portlists

msfschaffner, 2019-11-26:
The opentitan style guide https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md uses 2 spaces to indent the port lists of a module, whereas the current implementation of the formatter uses 4 spaces. 

E.g.
```systemverilog
import my_mod (
  input clk_i,
  input rst_ni
);
```
versus
```systemverilog
import my_mod (
    input clk_i,
    input rst_ni
);
```

We should discuss whether the OT style should align with 4 spaces, or whether the tool should support multiple indentation styles.

msfschaffner, 2019-11-26:
b/145168090

fangism, 2019-11-26:
Yes, please resolve style difference with your team first.  I hope the resolution is "just format it consistently for me, I don't care which way."  :)

msfschaffner, 2019-12-16:
I double checked and apparently we already had a lengthy discussion about this topic when creating the OT style guide, and there is no intent to change this. 

It would be nice if this style difference were configurable at some point. So far this is one of the only style elements we intend to keep...

I will open another issue to track these differences for convenience.

fangism, 2019-12-16:
My observation with clang-format (as configured for internal use) is that things inside `()` get wrapped 4 spaces and things inside `{}` get indented 2 spaces (and that is without any consideration of context).  

Generally agree that configurability is important to adoption in the long run, but not among the short term tasks.

msfschaffner, 2019-12-16:
Copying from [OT style guide](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md) for reference:

### Indentation

***Indentation is two spaces per level.***

Use spaces for indentation. Do not use tabs. You should set your editor to emit
spaces when you hit the tab key.

#### Indented Sections

Always add an additional level of indentation to the enclosed sections of all
paired keywords. Examples of SystemVerilog keyword pairs: `begin / end`,
`module / endmodule`, `package / endpackage`, `class / endclass`,
`function / endfunction`.

#### Line Wrapping

When wrapping a long expression, indent the continued part of the expression by
four spaces, like this:

&#x1f44d;
```systemverilog {.good}
assign zulu = enabled && (
    alpha < bravo &&
    charlie < delta);

assign addr = addr_gen_function_with_many_params(
    thing, other_thing, long_parameter_name, x, y,
    extra_param1, extra_param2);

assign structure = '{
    src: src,
    dest: dest,
    default: '0};
```

Or, if it improves readability, align the continued part of the expression with
a grouping open parenthesis or brace, like this:

```systemverilog
assign zulu = enabled && (alpha < bravo &&
                          charlie < delta);

assign addr = addr_gen_function(thing, other_thing,
                                long_parameter_name,
                                x, y);

assign structure = '{src: src,
                     dest: dest,
                     default: '0};
```

Operators in a wrapped expression can be placed at either the end or the
beginning of each line, but this must be done consistently within a file.

msfschaffner, 2019-12-16:
Would it make the configuration easier if we aligned the indentation in OT strictly to 2 spaces in every context? I.e. even when wrapping long expressions?

fangism, 2020-09-09:
After brief discussion with @msfschaffner , regarding the original request, we're specifically addressing the following contexts for now:

* port declarations
* named port connections
* parameter declarations
* named parameter assignments

closed 2020-09-10 by hzeller

fangism, 2020-09-10:
@msfschaffner @imphil @udinator

See these new (semi-documented) flags to configure indentation to OT's style in the short-term:

```
    --formal_parameters_indentation (Indent formal parameters: {indent,wrap});
      default: wrap;
    --named_parameter_indentation (Indent named parameter assignments:
      {indent,wrap}); default: wrap;
    --named_port_indentation (Indent named port connections: {indent,wrap});
      default: wrap;
    --port_declarations_indentation (Indent port declarations: {indent,wrap});
      default: wrap;
```

To get the 2 space indentation you want, set these all to `=indent`.

You can save these flags to a makeshift flag file and pass:

```
  Flags from absl/flags/parse.cc:
    --flagfile (comma-separated list of files to load flags from); default: ;
```

When these get re-organized in the future, we'll give you a head's up.

fangism, 2020-09-10:
Now available in release: https://github.com/google/verible/releases/tag/v0.0-579-gd6f06e4

msfschaffner, 2020-09-10:
Thanks @fangism! We'll give this a go.

msfschaffner, 2020-10-06:
I did a quick test and noted that the flags do not seem to be fully working when instantiating modules, see testcase below:

is:
```systemverilog
module bla;

  prim_alert_sender #(
      .AsyncOn(AlertAsyncOn[0])
  ) u_prim_alert_sender0 (
      .clk_i,
      .rst_ni,
      .alert_i   (otp_fatal_error),
      .alert_rx_i(alert_rx_i[0]),
      .alert_tx_o(alert_tx_o[0])
  );

endmodule
```

should:
```systemverilog
module bla;

  prim_alert_sender #(
    .AsyncOn(AlertAsyncOn[0])
  ) u_prim_alert_sender0 (
    .clk_i,
    .rst_ni,
    .alert_i   (otp_fatal_error),
    .alert_rx_i(alert_rx_i[0]),
    .alert_tx_o(alert_tx_o[0])
  );

endmodule
```

commit d6f06e4e (2020-09-10): Provide option to indent named parameters/ports 2 spaces instead of wrap +4.

fixes #40

PiperOrigin-RevId: 330980195
commit a1dc2bfc (2020-10-07): Propagate indentation flags through format_style.

Fixes #40

PiperOrigin-RevId: 335748540