chipsalliance/verible #42: [line breaks, wraps] Indentation/alignment issue inside `ASSUME macro

msfschaffner, 2019-11-26:
should:
```systemverilog
    `ASSERT(MioWarl_A,
            padctrl.reg2hw.mio_pads[mio_sel].qe |=>
        !(|padctrl.mio_attr_q[mio_sel][padctrl_reg_pkg::AttrDw-1:2]),
            clk_i, !rst_ni)
```
is:
```systemverilog
    `ASSERT(MioWarl_A,
            padctrl.reg2hw.mio_pads[mio_sel].qe |=>
        !(|padctrl.mio_attr_q[mio_sel][padctrl_reg_pkg::AttrDw-1:2])
            , clk_i, !rst_ni)
```

msfschaffner, 2019-11-26:
b/145170090

fangism, 2019-11-26:
I'd say the issue here is a poor-choice of breaking before the comma (before `clk_i`).  If so, that's a penalty tuning matter.

fangism, 2020-02-03:
Closer examination:

```systemverilog
module m;
initial begin
	    `ASSERT(MioWarl_A,
            padctrl.reg2hw.mio_pads[mio_sel].qe |=>
        !(|padctrl.mio_attr_q[mio_sel][padctrl_reg_pkg::AttrDw-1:2]),
            clk_i, !rst_ni)
    end
    endmodule
```

has this token-partition structure:

```
$ verilog_format --show_token_partition_tree ~/temp/verilog/formatting/b145170090.sv
Full token partition tree:
{ ([<auto>]) @{}, policy: always-expand
  { ([<auto>]) @{0}, policy: fit-else-expand
    { ([module m ;]) }
  }
  { (>>[<auto>]) @{1}, policy: always-expand
    { (>>[initial begin]) }
    { (>>>>[<auto>]) @{1,1}, policy: always-expand
      { (>>>>[<auto>]) @{1,1,0}, policy: fit-else-expand
        { (>>>>[`ASSERT (]) }
        { (>>>>>>>>[<auto>]) @{1,1,0,1}, policy: fit-else-expand
          { (>>>>>>>>[MioWarl_A , padctrl.reg2hw.mio_pads[mio_sel].qe |=>
        !(|padctrl.mio_attr_q[mio_sel][padctrl_reg_pkg::AttrDw-1:2]) , clk_i , ! rst_ni]) }
        }
        { (>>>>[)]) }
      }
    }
    { (>>[end]) }
  }
  { ([endmodule]) }
}
```

The macro call arguments are all lumped into the same partition, which is wrong; each argument should have its own partition.  Eventually, there should be a different policy for function/macro call arguments that compacts argument subpartitions onto the same line while they fit.  (The same policy is desired for function/task formal arguments in #29 ).

commit fd459f60 (2020-02-12): PR #175: Fix not sufficiently partitioned arguments in macro call
Fixes #42

GitHub PR https://github.com/google/verible/pull/175

Copybara import of the project:

  - c53ea7be4044a96dcbd903cbeb24329ecb059d18 Add AnalyzeVerilogPropertySpec() function by Lukasz Dalek <ldalek@antmicro.com>
  - ba19dd2aeeb607cbf8df9144705597327f66adb4 Try to parse macro arguments as property before applying ... by Lukasz Dalek <ldalek@antmicro.com>
  - e2d910e10c47cb26155ae6d4f86350e22ff6aa8e Fix MacroCall arguments partitioning by Lukasz Dalek <ldalek@antmicro.com>
  - 9be34faa566752a4db5ff39f9a65e905b3f47b22 Up [... cut]
commit 9f748a84 (2020-03-04): PR #197: Implement AppendFittingSubpartitions policy
Implements algorithm from https://github.com/google/verible/pull/148#issuecomment-581972774
Fixes #42 #29 and (?) #97

GitHub PR https://github.com/google/verible/pull/197

Copybara import of the project:

  - 682ffbbad6b88515589b219d736eeaae5e7b1437 Add current_column argument to FitsOnLine function by Lukasz Dalek <ldalek@antmicro.com>
  - 84ead8ade6ddc9db58ba8c130435f4ad0b1bc3a2 Add kAppendFittingSubPartitions policy by Lukasz Dalek <ldalek@antmicro.com>
  - ac2f83b41204c30e2626a0eabe4bbd30e7bb1921 Add transformation functions to tree_unw [... cut]
closed 2020-02-12 by hzeller