chipsalliance/verible #43: [space] Unneeded space in ternaries

msfschaffner, 2019-11-26:
should:
```systemverilog
  assign usb_tx_rready = sys_loopback ? lb_data_move : usb_if_tx_read;
  assign usb_rx_wvalid = sys_loopback ? lb_data_move : usb_if_rx_write;
  assign usb_rx_fifo_wdata = sys_loopback ? usb_tx_fifo_rdata : usb_if_rx_fifo_wdata;
```
is:
```systemverilog
  assign usb_tx_rready = sys_loopback ? lb_data_move      : usb_if_tx_read;
  assign usb_rx_wvalid = sys_loopback ? lb_data_move      : usb_if_rx_write;
  assign usb_rx_fifo_wdata = sys_loopback ? usb_tx_fifo_rdata : usb_if_rx_fifo_wdata;
```

msfschaffner, 2019-11-26:
b/145170233

fangism, 2019-11-26:
This case not handled, should be an easy fix.

Reduced test case:

```systemverilog
module m;
assign r = a ? b : c;
endmodule
```

```
$ verilog_format -v=4 --alsologtostderr  ~/testing/verilog/formatting/b145170233.sv
...
I1125 18:50:51.146008   96624 token_annotator.cc:114] Spacing between SymbolIdentifier and ':'
I1125 18:50:51.146015   96624 token_annotator.cc:315] spaces: -1, reason: Default: spacing not explicitly handled
I1125 18:50:51.146022   96624 token_annotator.cc:320] Unhandled inter-token spacing between SymbolIdentifier and ':', defaulting to 1
```

By default, 'unhandled' cases preserve original spacing, which explains why you still see multiple spaces.

commit 8296fd33 (2019-12-02): Handle spaces around ':' for a number of cases.

In particular, ternary expressions and statement/block labels.

fixes #43
fixes #48

PiperOrigin-RevId: 282659244
closed 2019-12-02 by hzeller