chipsalliance/verible #44: [keyword] Verible breaks when parsing a function that is defining “new”

msfschaffner, 2019-11-26:
The following code causes verible to throw an error and stop processing.

```systemverilog
`ifndef uvm_component_new
  `define uvm_component_new \
    function new (string name="", uvm_component parent=null); \
      super.new(name, parent); \
    endfunction : new
`endif
```

The associated error message is:
```
Mismatch in token sequence lengths: 693 vs. 744
First mismatched token [69]: ("<<`define-tokens>>") (#261: "\
    function new (string name=""); \
      super.new(name); \
    endfunction : new") vs. ("<<`define-tokens>>") (#261: "")
Formatted output is lexically different from the input.  Please file go/verilogformat-bug.
Error lex/parsing-ing formatted output.  Please file go/verilogformat-bug.
First error: token: "new" at 43:14-16
```

msfschaffner, 2019-11-26:
b/145170433

fangism, 2019-11-26:
This is a bug in the formatter's output verification logic.  And fail-safe behavior make it leave the original input untouched.

tgorochowik, 2020-03-18:
I think this issue may have been already fixed. I cannot reproduce it anymore, but it is easily reproducible by checking out code from November.

@msfschaffner could you please confirm that?

msfschaffner, 2020-04-01:
Confirmed, this particular instance does not seem to produce issues anymore. But I found another issue in the [same file](https://github.com/lowRISC/opentitan/blob/14ffb82597f9c29c254d36e5bdb40de5dfb6ee3f/hw/dv/sv/dv_utils/dv_macros.svh#L1-L288) (there are tons of DV macros in there). I'll spin that out as a separate issue.

fangism, 2020-04-01:
Thanks for the update and opening a new issue.

closed 2020-04-01 by fangism