chipsalliance/verible #58: Parse error at `input var`

dalance, 2019-11-28:
For example, the following module can't be parsed by verilog_syntax.

```
module A(
    input  var i,
    output var o
);

endmodule
```

```
$ verilog_syntax test.sv
test.sv:2:12: syntax error, rejected "var" (https://github.com/google/verible).
test.sv:6:1: syntax error, rejected "endmodule" (https://github.com/google/verible).
```

fangism, 2019-11-28:
```systemverilog
// highlighted with ```systemverilog
module A(
    input  var i,
    output var o
);
endmodule
```

```verilog
// highlighted with ```verilog
module A(
    input  var i,
    output var o
);
endmodule
```

In SystemVerilog `var` is a keyword, and cannot be used as a type.

Currently Verible only parses in SystemVerilog mode.

dalance, 2019-11-28:
I think `input var a` is valid in SystemVerilog.
The following BNF can take it.

```
input_declaration ::=
input net_port_type list_of_port_identifiers
| input variable_port_type list_of_variable_identifiers
variable_port_type ::= var_data_type
var_data_type ::= data_type | var data_type_or_implicit
data_type_or_implicit ::=
data_type
| implicit_data_type
implicit_data_type ::= [ signing ] { packed_dimension }
```

```
code: input var a
BNF : input variable_port_type list_of_variable_identifiers
code: var a
BNF : variable_port_type list_of_variable_identifiers
code: var a
BNF : var_data_type list_of_variable_identifiers
code: var a
BNF : var data_type_or_implicit list_of_variable_identifiers
code: a
BNF : data_type_or_implicit list_of_variable_identifiers
code: a
BNF : [ signing ] { packed_dimension } list_of_variable_identifiers
code: a
BNF : list_of_variable_identifiers
```

In LRM 6.8, there are some examples of `var`.
`input var a` is the same as `input var logic a`.

```systemverilog
var byte my_byte; // equivalent to "byte my_byte;"
var v; // equivalent to "var logic v;"
var [15:0] vw; // equivalent to "var logic [15:0] vw;"
var enum bit { clear, error } status;
input var logic data_in;
var reg r;
```

fangism, 2019-11-28:
Ok, thanks for that reference, I looked more closely, and here's the grammar reduction path I found through module declarations (very similar):

module_declaration, module_ansi_header, list_of_port_declarations, port_declaration, input_declaration, variable_port_type: 'var' data_type_or_implicit

So the Verible parser needs to be updated.

fangism, 2019-11-28:
b/145319501

mithro, 2019-11-28:
This world probably be a good test case to add to 
[sv-tests](https://GitHub.com/SymbiFlow/sv-tests).

fangism, 2019-12-03:
Fix is in the pipeline.

hzeller, 2019-12-03:
Filed https://github.com/SymbiFlow/sv-tests/issues/486 for inclusion in sv-tests.

commit 01b639c9 (2019-12-04): Parser: accept 'var' keyword in port declarations.

Added some ref-declaration-in-module-port unit-tests.

fixes #58

PiperOrigin-RevId: 283819064
closed 2019-12-04 by hzeller