chipsalliance/verible #65: Currently failing FormatterEndToEndTest.PenaltySensitiveLineWrapping test

mithro, 2019-12-03:
On git revision bbef19d99198973a6428aa283a0b48ca01bae10b it seems like the test FormatterEndToEndTest.PenaltySensitiveLineWrapping is failing.
```
//verilog/formatting:formatter_test                                      FAILED in 0.3s
  /usr/local/google/home/tansell/.cache/bazel/_bazel_tansell/d19eda4ce6404db62d974dacbe6d8f5f/execroot/com_google_verible/bazel-out/k8-fastbuild/testlogs/verilog/formatting/formatter_test/test.log

Executed 1 out of 295 tests: 294 tests pass and 1 fails locally.
INFO: Build completed, 1 test FAILED, 2 total actions
```
See output below;
```
[ RUN      ] FormatterEndToEndTest.PenaltySensitiveLineWrapping
verilog/formatting/formatter_test.cc:1830: Failure
Expected equality of these values:
  stream.str()
    Which is: "module m;\n  assign wwwwww[77 : 66]\n      = sss(qqqq[33 : 22], vv[44 : 1]);\nendmodule\n"
  test_case.expected
    Which is: { 'm' (109, 0x6D), 'o' (111, 0x6F), 'd' (100, 0x64), 'u' (117, 0x75), 'l' (108, 0x6C), 'e' (101, 0x65), ' ' (32, 0x20), 'm' (109, 0x6D), ';' (59, 0x3B), '\n' (10, 0xA), ' ' (32, 0x20), ' ' (32, 0x20), 'a' (97, 0x61), 's' (115, 0x73), 's' (115, 0x73), 'i' (105, 0x69), 'g' (103, 0x67), 'n' (110, 0x6E), ' ' (32, 0x20), 'w' (119, 0x77), 'w' (119, 0x77), 'w' (119, 0x77), 'w' (119, 0x77), 'w' (119, 0x77), 'w' (119, 0x77), '[' (91, 0x5B), '7' (55, 0x37), '7' (55, 0x37), ' ' (32, 0x20), ':' (58, 0x3A), ' ' (32, 0x20), '6' (54, 0x36), ... }
With diff:
@@ -1,4 +1,2 @@
-module m;
-  assign wwwwww[77 : 66]
-      = sss(qqqq[33 : 22], vv[44 : 1]);
-endmodule\n
+{ 'm' (109, 0x6D), 'o' (111, 0x6F), 'd' (100, 0x64), 'u' (117, 0x75), 'l' (108, 0x6C), 'e' (101, 0x65), ' ' (32, 0x20), 'm' (109, 0x6D), ';' (59, 0x3B), '
+' (10, 0xA), ' ' (32, 0x20), ' ' (32, 0x20), 'a' (97, 0x61), 's' (115, 0x73), 's' (115, 0x73), 'i' (105, 0x69), 'g' (103, 0x67), 'n' (110, 0x6E), ' ' (32, 0x20), 'w' (119, 0x77), 'w' (119, 0x77), 'w' (119, 0x77), 'w' (119, 0x77), 'w' (119, 0x77), 'w' (119, 0x77), '[' (91, 0x5B), '7' (55, 0x37), '7' (55, 0x37), ' ' (32, 0x20), ':' (58, 0x3A), ' ' (32, 0x20), '6' (54, 0x36), ... }

code:
module m;assign wwwwww[77:66]= sss(qqqq[33:22],vv[44:1]);endmodule
[  FAILED  ] FormatterEndToEndTest.PenaltySensitiveLineWrapping (1 ms)
```

fangism, 2019-12-03:
@hzeller already has a 'fix' coming.

Explanation: to-date, there has been zero effort put into selecting sensible penalty values in the formatters wrap-optimization.  There are situations where the best (minimum) penalty is reached via more than one solution/path.  In such cases, the selection of the winner is subject to the implementation details of the underlying heap implementation of std::priority_queue.

Here is where penalties are assigned:

https://github.com/google/verible/blob/bbef19d99198973a6428aa283a0b48ca01bae10b/verilog/formatting/token_annotator.cc#L413

Here is debug code to demonstrate:

https://github.com/google/verible/blob/bbef19d99198973a6428aa283a0b48ca01bae10b/common/formatting/line_wrap_searcher.cc#L135

We could prevent this problem by requiring that penalty-tuning tests have one unique solution.

b/145558510

hzeller, 2019-12-03:
Fixed with https://github.com/google/verible/commit/8a8c341b82ec5c1ba63b7c82f52da081a4b239f6

closed 2019-12-03 by hzeller