chipsalliance/verible #69: Thoughts on future directions

MikePopoloski, 2019-12-03:
Hey guys, this isn't really an issue but more of an attempt to "contact you". I think it's great that more people are working on Verilog tools, and I like that there's a desire for a common front end library that can be used by anyone. I've been working on such a project myself for a while now and it's quite a bit further along; perhaps it makes sense to combine efforts instead of duplicating the work? The project is here: https://github.com/MikePopoloski/slang

My parser is a hand written recursive descent parser, like nearly all production grade compilers end up using at some point, with the goal of producing great error messages that are difficult or impossible to produce from flex/bison generated parsers. Additionally I've done a pretty significant chunk of the work of doing type checking, compile time evaluation, and elaboration, which is a pretty huge effort if you end up wanting to tackle that as well. Finally, I'm planning soon to start integrating LLVM as a backend to produce simulation binaries, though the whole project is designed as a modular set of libraries that can be linked into any tool, such as a synthesis project like Yosys.

If you feel like you want to continue on with your own project I understand and there will be no hard feelings, but it certainly seems like we could accomplish a lot more by joining efforts and slang as a platform is much further along. Thanks for your time!

fangism, 2019-12-03:
Hi Mike, nice to meet the slang lead!  This would be great to discuss on our general dev list, please join us:
https://groups.google.com/forum/#!forum/verible-dev

I think there is tremendous collaborative opportunity on front-end tools.  Verible's main purpose was supporting operations on **unpreprocessed** standalone source files (style linting and formatting), which is what distinguishes it from production compilation toolchains (like what separates some clang-tools from clang's proper front-end).  It doesn't even have an AST today, because it doesn't need one yet.  I actually have very little expertise in Verilog-specific matters and semantics and elaboration, and that is where we would welcome help and collaboration.  Syntax diagnostics are a known weakness today, but I do know how to leverage some of bison's internals to improve that situation.  Whether Verible could open a pathway to a proper front-end for SV is an open question, let alone whether it is among the better options.  (Merely stripping out everything preprocessor-related from our verilog.y would give a good head-start.)  

Some good topics for our teams to discuss:
* test suites for standards conformance.  https://github.com/SymbiFlow/sv-tests is collecting tests, I believe slang fares quite well.
* implementation-neutral representations, including ASTs, parsed and elaborated forms.

So indeed, no hard feelings, let's continue to promote collaborative discussion.

MikePopoloski, 2019-12-04:
Style linting and formatting was actually also a goal of slang's parser; the "parse tree" that it produces (separate from the AST) is designed such that it always fully round trips back to the original source, comments and whitespace and all, no matter how malformed the input text is. If you're familiar with the C# Roslyn compiler's design, this borrows heavily from that. The idea being that once you're in parse tree form, you can easily write refactoring tools that move syntax nodes around and all of the comments and whitespace come along for the ride. To be clear not much work has been done to make use of that functionality, but modulo any bugs it should work. The "rewriter" tool in my repository is designed as a test case for that, where it just parses into a tree and then roundtrips it back to source code, which you should be able to compare byte-for-byte for correctness.

I joined your Google group, looking forward to following along with any discussions that may follow (and I'll close this issue since that seems like a better medium for back-and-forth).

closed 2019-12-04 by MikePopoloski