chipsalliance/verible #72: Tests currently failing?

mithro, 2019-12-03:
https://travis-ci.org/google/verible/jobs/619947264
```
//verilog/tools/lint:void-cast-onerule-test                              PASSED in 0.1s
//verilog/tools/lint:void-cast-syntax-test                               PASSED in 0.1s
//common/strings:rebase_test                                             FAILED in 0.3s
  /home/travis/.cache/bazel/_bazel_travis/09cbf2eb38e4f867e4fa81d2eb791463/execroot/com_google_verible/bazel-out/k8-fastbuild/testlogs/common/strings/rebase_test/test.log
//common/text:token_info_test                                            FAILED in 0.3s
  /home/travis/.cache/bazel/_bazel_travis/09cbf2eb38e4f867e4fa81d2eb791463/execroot/com_google_verible/bazel-out/k8-fastbuild/testlogs/common/text/token_info_test/test.log
//verilog/analysis:verilog_linter_configuration_test                     FAILED in 0.2s
  /home/travis/.cache/bazel/_bazel_travis/09cbf2eb38e4f867e4fa81d2eb791463/execroot/com_google_verible/bazel-out/k8-fastbuild/testlogs/verilog/analysis/verilog_linter_configuration_test/test.log

Executed 295 out of 295 tests: 292 tests pass and 3 fail locally.
```

mithro, 2019-12-03:
This seems to be failing on both Travis and Kokoro.

https://source.cloud.google.com/results/invocations/873600c2-66ae-458f-85f2-d9ba507e8d29/targets

hzeller, 2019-12-03:
I have seen the verilog linter configuration fail with gcc 9 (issue #66), but should work fine with gcc-8 (having said that we should of course make sure that it works with gcc-9 as well).

The rebase_test and token_info_test I have not seen fail before locally neither with gcc-8 or gcc-9, I'll have to inspect the logs in travis. Is it possible to click through to that log-file content ?

closed 2020-01-08 by hzeller