chipsalliance/verible #73: CI: use --noshow_progress on bazel builds

hzeller, 2019-12-03:
bazel outputs a continuous stream of percentage of how far it got in the build-process. In the CI case, this means that the build log is pretty large and not super-useful.

So for all CI invocations, we should probably give the `--noshow_progress` option to blaze:
```
bazel test --noshow_progress --cxxopt='-std=c++17' //...
```

mithro, 2019-12-03:
FYI - Other progress related command line arguments seem to be;
```
  --progress_report_interval (an integer in 0-3600 range; default: "0")
  --[no]progress_in_terminal_title (a boolean; default: "false")
  --[no]show_loading_progress (a boolean; default: "true")
  --[no]show_progress (a boolean; default: "true")
  --show_progress_rate_limit (a double; default: "0.2")
```

mithro, 2019-12-03:
There is also;
```
  --test_output (summary, errors, all or streamed; default: "summary")
```

mithro, 2019-12-03:
I think we might want;
[`--attempt_to_print_relative_paths`](https://docs.bazel.build/versions/master/command-line-reference.html#flag--attempt_to_print_relative_paths)

mithro, 2019-12-03:
Actually, we should probably look at all the `tag: terminal_output` command line arguments.

commit 3ac656b8 (2019-12-03): kokoro: Don't show bazel progress.

Fixes #73.
PR #75 (2019-12-03): travis: Small fixes
 * Don't output progress - Fixes #73 
 * Include log files in the output on failure.
commit bbb0c229 (2019-12-03): travis: Don't output bazel progress.

Fixes #73.
closed 2019-12-03 by mithro