chipsalliance/verible #78: Build failure on RHEL/CentOS7

dalance, 2019-12-04:
I tried to build on RHEL/CentOS 7 environment, but the build was failed.
The environment can reproduce by the following Dockerfile.

```Dockerfile
FROM centos:7

ADD https://copr.fedorainfracloud.org/coprs/vbatts/bazel/repo/epel-7/vbatts-bazel-epel-7.repo /etc/yum.repos.d

RUN yum install -y centos-release-scl
RUN yum install -y \
        devtoolset-8 \
        git \
        wget \
        bazel

SHELL [ "scl", "enable", "devtoolset-8"]

RUN git clone https://github.com/google/verible

RUN gcc --version; cd verible; bazel build --cxxopt='-std=c++17' //...
```

The execution result is below:

```
Sending build context to Docker daemon  4.608kB
Step 1/7 : FROM centos:7
 ---> 1e1148e4cc2c
Step 2/7 : ADD https://copr.fedorainfracloud.org/coprs/vbatts/bazel/repo/epel-7/vbatts-bazel-epel-7.repo /etc/yum.repos.d
Downloading [==================================================>]     346B/346B
 ---> Using cache
 ---> 8efe2fe1f84a
Step 3/7 : RUN yum install -y centos-release-scl
 ---> Using cache
 ---> 1aa26361b752
Step 4/7 : RUN yum install -y         devtoolset-8         git         wget         bazel
 ---> Using cache
 ---> 9f461b9bc390
Step 5/7 : SHELL [ "scl", "enable", "devtoolset-8"]
 ---> Using cache
 ---> b9429cb6000f
Step 6/7 : RUN git clone https://github.com/google/verible
 ---> Running in 017886f07552
Cloning into 'verible'...
Removing intermediate container 017886f07552
 ---> 1137b30fcfae
Step 7/7 : RUN gcc --version; cd verible; bazel build --cxxopt='-std=c++17' //...
 ---> Running in 19a98491345c
gcc (GCC) 8.3.1 20190311 (Red Hat 8.3.1-3)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Extracting Bazel installation...
Starting local Bazel server and connecting to it...
Loading:
Loading: 0 packages loaded
Loading: 0 packages loaded
Loading: 18 packages loaded
    currently loading: bazel
Analyzing: 832 targets (19 packages loaded, 0 targets configured)
Analyzing: 832 targets (39 packages loaded, 309 targets configured)
Analyzing: 832 targets (58 packages loaded, 1340 targets configured)
Analyzing: 832 targets (59 packages loaded, 1671 targets configured)
Analyzing: 832 targets (59 packages loaded, 1671 targets configured)
Analyzing: 832 targets (60 packages loaded, 2846 targets configured)
Analyzing: 832 targets (61 packages loaded, 2875 targets configured)
Analyzing: 832 targets (62 packages loaded, 4145 targets configured)
Analyzing: 832 targets (63 packages loaded, 4263 targets configured)
INFO: Analyzed 832 targets (63 packages loaded, 4558 targets configured).
INFO: Found 832 targets...
[0 / 150] [Prepa] Writing file external/com_google_googletest/libgtest.so-2.params ... (2 actions, 0 running)
[149 / 238] Compiling external/com_google_googletest/googletest/src/gtest.cc; 2s processwrapper-sandbox ... (32 actions, 31 running)
[239 / 432] CcConfigureMakeRule bazel/m4/include [for host]; 2s processwrapper-sandbox ... (32 actions, 31 running)
ERROR: /verible/common/text/BUILD:79:1: Linking of rule '//common/text:concrete_syntax_leaf_test' failed (Exit 1) gcc failed: error executing command /opt/rh/devtoolset-8/root/usr/bin/gcc @bazel-out/k8-fastbuild/bin/common/text/concrete_syntax_leaf_test-2.params

Use --sandbox_debug to see verbose messages from the sandbox
bazel-out/k8-fastbuild/bin/_solib_k8/libcommon_Stext_Slibtoken_Uinfo.so: error: undefined reference to 'std::__throw_out_of_range_fmt(char const*, ...)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::logic_error::logic_error(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::invalid_argument::invalid_argument(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::domain_error::domain_error(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::length_error::length_error(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::out_of_range::out_of_range(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::runtime_error::runtime_error(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::range_error::range_error(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::overflow_error::overflow_error(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Uabsl_Sabsl_Sbase_Slibthrow_Udelegate.so: error: undefined reference to 'std::underflow_error::underflow_error(char const*)'
bazel-out/k8-fastbuild/bin/_solib_k8/libexternal_Scom_Ugoogle_Ugoogletest_Slibgtest.so: error: undefined reference to 'std::runtime_error::runtime_error(char const*)'
bazel-out/k8-fastbuild/bin/common/text/_objs/concrete_syntax_leaf_test/concrete_syntax_leaf_test.pic.o:concrete_syntax_leaf_test.cc:function verible::(anonymous namespace)::ValueSymbolTest_EqualityArity2Args_Test::~ValueSymbolTest_EqualityArity2Args_Test(): error: undefined reference to 'operator delete(void*, unsigned long)'
bazel-out/k8-fastbuild/bin/common/text/_objs/concrete_syntax_leaf_test/concrete_syntax_leaf_test.pic.o:concrete_syntax_leaf_test.cc:function testing::internal::TestFactoryImpl<verible::(anonymous namespace)::ValueSymbolTest_EqualityArity2Args_Test>::~TestFactoryImpl(): error: undefined reference to 'operator delete(void*, unsigned long)'
bazel-out/k8-fastbuild/bin/common/text/_objs/concrete_syntax_leaf_test/concrete_syntax_leaf_test.pic.o:concrete_syntax_leaf_test.cc:function verible::(anonymous namespace)::ValueSymbolTest_EqualityArity1Args_Test::~ValueSymbolTest_EqualityArity1Args_Test(): error: undefined reference to 'operator delete(void*, unsigned long)'
bazel-out/k8-fastbuild/bin/common/text/_objs/concrete_syntax_leaf_test/concrete_syntax_leaf_test.pic.o:concrete_syntax_leaf_test.cc:function testing::internal::TestFactoryImpl<verible::(anonymous namespace)::ValueSymbolTest_EqualityArity1Args_Test>::~TestFactoryImpl(): error: undefined reference to 'operator delete(void*, unsigned long)'
collect2: error: ld returned 1 exit status
INFO: Elapsed time: 21.035s, Critical Path: 5.14s
INFO: 166 processes: 166 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
FAILED: Build did NOT complete successfully
The command 'scl enable devtoolset-8 gcc --version; cd verible; bazel build --cxxopt='-std=c++17' //...' returned a non-zero code: 1
```

I tried `devtoolset-7` ( gcc 7.3.1 ) too. It failed at the similar error.

fangism, 2019-12-04:
Those symbols look like libstdc++ symbols.  Can you verify that libstdc++ is installed on the host?  There should be a version that matches your g++ version.

dalance, 2019-12-04:
Thanks. The version of libstdc++ is the same as g++.
This issue may be related to bazel's issue https://github.com/bazelbuild/bazel/issues/10327.

dalance, 2019-12-04:
I tried some workarounds.

```
RUN gcc --version; cd verible; BAZEL_LINKOPTS="-static-libstdc++:-lm" BAZEL_LINKLIBS="-l%:libstdc++.a" bazel build --sandbox_debug --cxxopt='-std=c++17' //...
```

The link error was resolved, but another compile error happened.

```
[0 / 93] [Prepa] Creating source manifest for //common/parser:move_yacc_stack_symbols
[119 / 519] Compiling external/com_google_googletest/googlemock/src/gmock-internal-utils.cc; 2s processwrapper-sandbox ... (32 actions, 31 running)
[224 / 534] Compiling external/com_google_googletest/googletest/src/gtest.cc; 5s processwrapper-sandbox ... (32 actions, 31 running)
[505 / 803] CcConfigureMakeRule bazel/m4/include [for host]; 3s processwrapper-sandbox ... (32 actions, 31 running)
[1,197 / 1,700] CcConfigureMakeRule bazel/m4/include [for host]; 7s processwrapper-sandbox ... (32 actions, 31 running)
ERROR: /verible/common/formatting/BUILD:242:1: C++ compilation of rule '//common/formatting:line_wrap_searcher_test' failed (Exit 1) process-wrapper failed: error executing command
  (cd /root/.cache/bazel/_bazel_root/74978e9359729ba3edce57df66847b47/sandbox/processwrapper-sandbox/486/execroot/com_google_verible && \
  exec env - \
    LD_LIBRARY_PATH=/opt/rh/devtoolset-8/root/usr/lib64:/opt/rh/devtoolset-8/root/usr/lib:/opt/rh/devtoolset-8/root/usr/lib64/dyninst:/opt/rh/devtoolset-8/root/usr/lib/dyninst:/opt/rh/devtoolset-8/root/usr/lib64:/opt/rh/devtoolset-8/root/usr/lib \
    PATH=/opt/rh/devtoolset-8/root/usr/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin \
    PWD=/proc/self/cwd \
    TMPDIR=/tmp \
  /root/.cache/bazel/_bazel_root/install/be94b975a405b1ff4c8f3e1c55c94af6/process-wrapper '--timeout=0' '--kill_delay=15' /opt/rh/devtoolset-8/root/usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++0x' -MD -MF bazel-out/k8-fastbuild/bin/common/formatting/_objs/line_wrap_searcher_test/line_wrap_searcher_test.pic.d '-frandom-seed=bazel-out/k8-fastbuild/bin/common/formatting/_objs/line_wrap_searcher_test/line_wrap_searcher_test.pic.o' -fPIC -D__CLANG_SUPPORT_DYN_ANNOTATION__ -iquote . -iquote bazel-out/k8-fastbuild/bin -iquote external/com_google_absl -iquote bazel-out/k8-fastbuild/bin/external/com_google_absl -iquote external/com_google_googletest -iquote bazel-out/k8-fastbuild/bin/external/com_google_googletest -iquote external/bazel_tools -iquote bazel-out/k8-fastbuild/bin/external/bazel_tools -isystem external/com_google_googletest/googlemock -isystem bazel-out/k8-fastbuild/bin/external/com_google_googletest/googlemock -isystem external/com_google_googletest/googlemock/include -isystem bazel-out/k8-fastbuild/bin/external/com_google_googletest/googlemock/include -isystem external/com_google_googletest/googletest -isystem bazel-out/k8-fastbuild/bin/external/com_google_googletest/googletest -isystem external/com_google_googletest/googletest/include -isystem bazel-out/k8-fastbuild/bin/external/com_google_googletest/googletest/include '-std=c++17' -fno-canonical-system-headers -Wno-builtin-macro-redefined '-D__DATE__="redacted"' '-D__TIMESTAMP__="redacted"' '-D__TIME__="redacted"' -c common/formatting/line_wrap_searcher_test.cc -o bazel-out/k8-fastbuild/bin/common/formatting/_objs/line_wrap_searcher_test/line_wrap_searcher_test.pic.o)
In file included from external/com_google_googletest/googletest/include/gtest/gtest.h:62,
                 from external/com_google_googletest/googlemock/include/gmock/internal/gmock-internal-utils.h:47,
                 from external/com_google_googletest/googlemock/include/gmock/gmock-actions.h:51,
                 from external/com_google_googletest/googlemock/include/gmock/gmock.h:59,
                 from common/formatting/line_wrap_searcher_test.cc:19:
common/formatting/line_wrap_searcher_test.cc: In member function 'virtual void verible::{anonymous}::SearchLineWrapsTestFixture_DisplayEquallyOptimalWrappings_Test::TestBody()':
common/formatting/line_wrap_searcher_test.cc:357:21: error: 'StrContains' is not a member of 'absl'
   EXPECT_TRUE(absl::StrContains(stream.str(), "Found 2 equally good"));
                     ^~~~~~~~~~~
common/formatting/line_wrap_searcher_test.cc:358:21: error: 'StrContains' is not a member of 'absl'
   EXPECT_TRUE(absl::StrContains(stream.str(), "============"));
                     ^~~~~~~~~~~
INFO: Elapsed time: 37.080s, Critical Path: 11.38s
INFO: 520 processes: 520 processwrapper-sandbox.
FAILED: Build did NOT complete successfully
FAILED: Build did NOT complete successfully
```

If it is bazel's issue, I'll wait until it is fixed.

BTW, verible is not used on RHEL/CentOS7?
In my understanding, RHEL/CentOS7 is the most major platform supported by commercial EDA vendors.

fangism, 2019-12-04:
Long story short, Verible grew up in a non-EDA-tool-friendly, non-RHEL/Centos environment, and is only recently being tested against 'foreign' environments and toolchains.  Bear with us, as we shake out these early portability issues.  The compile error you see is probably attributed to a symbol that is available in our internal version of the absl library but not yet released.

fangism, 2019-12-04:
Ok, I have a fix for `absl::StrContains` in the pipeline.  Expect it to land some time tomorrow, after it is (manually) pushed to master.

dalance, 2019-12-04:
OK. I understand the current status. I'll try the next revision tomorrow.

mithro, 2019-12-04:
Looks like the `absl::StrContains` issue was replicated on Ubuntu at https://travis-ci.com/SymbiFlow/conda-packages/jobs/262914281

fangism, 2019-12-04:
This is a case where internally, we get a dependency transitively, but externally we do not, due to outbound transformations.  We have tools to detect and fix this.  We also have yet to connect internal presubmits to external presubmits.

dalance, 2019-12-05:
The latest revision was successfully compiled.
Thank you.

The final Dockerfile becomes below:

```Dockerfile
FROM centos:7

ADD https://copr.fedorainfracloud.org/coprs/vbatts/bazel/repo/epel-7/vbatts-bazel-epel-7.repo /etc/yum.repos.d

RUN yum install -y centos-release-scl
RUN yum install -y \
        devtoolset-8 \
        git \
        wget \
        bazel

SHELL [ "scl", "enable", "devtoolset-8"]

RUN git clone https://github.com/google/verible

RUN gcc --version; cd verible; BAZEL_LINKOPTS="-static-libstdc++:-lm" BAZEL_LINKLIBS="-l%:libstdc++.a" bazel build --cxxopt='-std=c++17' //...
```

fangism, 2019-12-05:
great to hear!
@hzeller @mithro Would it be appropriate to include a Dockerfile in our source tree and add this to our CI?

mithro, 2019-12-05:
I think having docker builds seems like a good idea. Only issue is there is too many options to choose from here! :-P - The options as I see them are;
 - Use Travis CI - https://docs.travis-ci.com/user/docker/#building-a-docker-image-from-a-dockerfile
 - Use Google Cloud Build - https://cloud.google.com/cloud-build/docs/quickstart-docker
 - Use Docker Hub - https://docs.docker.com/docker-hub/builds/

Google Cloud Build is probably the fastest option. Travis CI is probably the most usable by people who fork the repository.

mithro, 2020-02-12:
The pull request at #168 should provide prebuilt binaries for;
 - Centos 6, 7 and 8
 - Ubuntu LTS version (Trusty 14.04, Xenial 16.04, Bionic 18.04)
 - Ubuntu latest version (Eoan 19.10)

closed 2020-02-12 by mithro