chipsalliance/verible #81: Parse `alias` construct

fangism, 2019-12-07:
`alias` constructs are currently not parsed.

Task:
update grammar and tests:
https://github.com/google/verible/blob/master/verilog/parser/verilog.y
https://github.com/google/verible/blob/master/verilog/parser/verilog_parser_unittest.cc

Add a new node enum to:
https://github.com/google/verible/blob/master/verilog/CST/verilog_nonterminals.h

Expect this to be fairly easy.

fangism, 2019-12-07:
b/135705449

hzeller, 2019-12-07:
This is 10.11 in the LRM, which is not covered yet in the https://github.com/SymbiFlow/sv-tests - might be worthwhile to add there as well to have as compliance test ?

fangism, 2019-12-07:
Of course, feel free to copy unit tests in either direction!

commit 4ce910cd (2020-02-20): PR #211: Parse `alias` construct
Fixes #81

GitHub PR https://github.com/google/verible/pull/211

Copybara import of the project:

  - 182ceb77b1aeced4554737884001391ced5e752f Parse net alias construct by Tomasz Gorochowik <tgorochowik@antmicro.com>
  - 59643f68afa09282337ef4ae671cd837adfb3e6a Add net alias consturct tests by Tomasz Gorochowik <tgorochowik@antmicro.com>

Closes #211

PiperOrigin-RevId: 296280624
closed 2020-02-20 by hzeller