chipsalliance/verible #100: [line break] Function with one single argument

msfschaffner, 2019-12-16:
is:
```systemverilog
function automatic void set_max_quit_count(
    int n
);
```
should:
```systemverilog
function automatic void set_max_quit_count(int n);
```

msfschaffner, 2019-12-16:
b/146355412

fangism, 2019-12-16:
I get:

```systemverilog
function automatic void set_max_quit_count(int n);
endfunction
```

Can you include more surrounding context from your original test case?

msfschaffner, 2019-12-17:
looks like this only appears when a comment is placed on the previous line:
```systemverilog
  // Simple function to set max errors before quitting sim
  function automatic void set_max_quit_count(
      int n
  );
```

fangism, 2020-01-29:
I've confirmed this is already fixed:

```systemverilog
// loooooooooooooooooooooooooooooooo comment.................
function automatic void set_max_quit_count(
	    int n
    );
    endfunction
```

becomes:

```systemverilog
// loooooooooooooooooooooooooooooooo comment.................
function automatic void set_max_quit_count(int n);
endfunction
```

closed 2020-01-29 by fangism