chipsalliance/verible #101: [spaces] Space after streaming operator

msfschaffner, 2019-12-16:
Not sure whether we want to fix this, but we usually write
```systemverilog
return {<<8{data}};
```
instead of using a space after `<<` as inserted by Verible:
```systemverilog
return {<< 8{data}};
```

msfschaffner, 2019-12-16:
b/146356297

fangism, 2019-12-16:
Confirmed.

Debugging `verilog_format -v=4 --alsologtostderr` shows:

```
I1216 13:26:44.401413   53552 token_annotator.cc:121] Spacing between "<<" and TK_DecNumber
I1216 13:26:44.401417   53552 token_annotator.cc:385] spaces: 1, reason: Space around binary and assignment operators
```

It currently fails to account for context that `<<` is not a binary operator in this context:

```
$ verilog_syntax --printtree ~/testing/verilog/formatting/b146356297.sv

Parse Tree: 
Node (tag: kDescriptionList) {
  Node (tag: kFunctionDeclaration) {
    Node (tag: kFunctionHeader) {
      (#"function" @0-8: "function")
      Node (tag: kDataType) {
      }
      Node (tag: kUnqualifiedId) {
        (#SymbolIdentifier @9-10: "f")
      }
      (#';' @10-11: ";")
    }
    Node (tag: kStatementList) {
      Node (tag: kJumpStatement) {
        (#"return" @13-19: "return")
        Node (tag: kExpression) {
          Node (tag: kExpression) {
            Node (tag: kStreamingConcatenation) {      // immediate parent context
              (#'{' @20-21: "{")
              (#"<<" @21-23: "<<")                     // operator in question
              Node (tag: kNumber) {
                (#TK_DecNumber @23-24: "8")
              }
              Node (tag: kBraceGroup) {
                (#'{' @24-25: "{")
                Node (tag: kStreamExpressionList) {
                  Node (tag: kExpression) {
                    Node (tag: kReferenceCallBase) {
                      Node (tag: kReference) {
                        Node (tag: kLocalRoot) {
                          Node (tag: kUnqualifiedId) {
                            (#SymbolIdentifier @25-29: "data")
                          }
                        }
                      }
                    }
                  }
                }
                (#'}' @29-30: "}")
              }
              (#'}' @30-31: "}")
            }
          }
        }
        (#';' @31-32: ";")
      }
    }
    (#"endfunction" @33-44: "endfunction")
  }
}
```

commit 9ca0b03a (2020-01-13): PR #117: Fix spaces around streaming operator
Fixes #101.

GitHub PR https://github.com/google/verible/pull/117

Copybara import of the project:

  - a8b416e9257222f1034f55de1dcc715ec54a02ae Avoid spaces around streaming operators by Rafal Kapuscik <rkapuscik@antmicro.com>
  - e620855c09521219d83052f80f4198c4710fd2b3 Add token_annotator tests by Rafal Kapuscik <rkapuscik@antmicro.com>

Closes #117

PiperOrigin-RevId: 289496662
closed 2020-01-13 by hzeller