chipsalliance/verible #93: [space] Redundant space in bitslice

msfschaffner, 2019-12-16:
is:
 ```systemverilog
 always_ff @(posedge clk_i) begin
    automatic byte p2d = spidpi_tick(ctx, d2p);
    spi_device_sck_o <= p2d[0];
    spi_device_csb_o <= p2d[1];
    spi_device_mosi_o <= p2d[2];
    // stop verilator warning
    unused_dummy <= |p2d[7: 3];
  end
```

I expect the bitslice in the last non-blocking assignment to be in either of the two following formats:
```systemverilog
unused_dummy <= |p2d[7:3];
unused_dummy <= |p2d[7 : 3];
```

msfschaffner, 2019-12-16:
b/146348359

fangism, 2019-12-16:
Acknowledged.  Currently, we make a choice to leave spacing inside `[]` alone in many contexts, but we could limit that further if there was a firm decision on spacing around ranges.  It hasn't been clear when the space padding around `:` is desired.

rkapuscik, 2020-01-14:
I have implemented the corresponding rule, but please confirm how many spaces are desired:
* the style guide shows [no spaces around the `:` ](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#bit-slicing), but does not have an explicit rule
* current unit tests default to [one space](https://github.com/google/verible/blob/master/verilog/formatting/formatter_test.cc#L2451)
* `+:` and `-:` are treated as binary operators and default to one space

fangism, 2020-01-14:
@msfschaffner @snsokolov 
It still hasn't been decided whether these index-like contexts (or dimensions) want 0 or 1 space, so the current implementation leaves it alone, intentionally.  In spirit, we allow compaction inside indices, 0 spaces, as an exception to the binary operator guideline.  I think we are in agreement however that:
1) spaces should be limited to no more than 1 (except, perhaps, when we eventually implement alignment)
2) unbalanced spaces on both side of `:` (and similar separators) is undesirable.  

I think a non-controversial step would limit spaces to 1, but permit 0, and *symmetrize* the spacing so the spacing on the right of `:` follows the spacing on the left.

Thoughts from the others?

msfschaffner, 2020-01-14:
Hey @fangism, your suggestion makes a lot of sense to me. limit to 1, permit 0 and make sure the format is symmetric.

fangism, 2020-01-14:
Thanks, @msfschaffner .  Let's move forward with that plan.  If possible, apply this policy to a number of similar looking contexts (inside `[]`): array declarations, indexing, slicing.

commit 005da28a (2020-01-16): PR #126: Symmetrize spacing in ranges
Fixes #93.

GitHub PR https://github.com/google/verible/pull/126

Copybara import of the project:

  - e62c6a7d896e0ad0e09570524dd89b6be1d35118 Add rules for spacing in bit slice by Rafal Kapuscik <rkapuscik@antmicro.com>
  - def8713b5b90121dde9d1a37a1893809f7eb004b Add rules for ranges, symmetrize spaces by Rafal Kapuscik <rkapuscik@antmicro.com>
  - a7610195512a694a09deed0839c6a1cc7522027c Add more unit tests by Rafal Kapuscik <rkapuscik@antmicro.com>

Closes #126

PiperOrigin-RevId: 290155393
closed 2020-01-16 by hzeller