chipsalliance/verible #94: [line break/wrap] Erroneous line wrap after $urandom_range

msfschaffner, 2019-12-16:
There should be a line break after `;`

is:
```systemverilog
wr_data = $urandom_range((1 << intr_state_csrs[test_index[i]].get_n_used_bits()) -
                         1); exp_intr_state[test_index[i]] &= (~wr_data);
```
should:
```systemverilog
wr_data = $urandom_range((1 << intr_state_csrs[test_index[i]].get_n_used_bits()) - 1);
exp_intr_state[test_index[i]] &= (~wr_data);
```

msfschaffner, 2019-12-16:
b/146351585

fangism, 2019-12-16:
Confirmed.

Smaller complete test case:

```systemverilog
function f;
  wr_data = $urandom_range(); exp_intr_state[i] &= (~wr_data);
endfunction
```

`verilog_format --show_token_partition_tree` shows:

```
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([function f ;]) }
  }
  { (>>[<auto>]) @{1}
    { (>>[wr_data = $urandom_range ( ) ; exp_intr_state [ i ] &= ( ~ wr_data ) ;]) }
  }
  { ([endfunction]) }
}
```

The assignment statements should be in their own partitions.

Around here:
https://github.com/google/verible/blob/78170927a2f881a6189a71a871b62701eb6ddfa0/verilog/formatting/tree_unwrapper.cc#L646

We're simply missing a case for `kAssignModifyStatement` (evident when running `verilog_syntax --printtree`).

commit e2e28a86 (2020-01-17): PR #131: Add partitioning around kAssignModifyStatement
Fixes #94.

GitHub PR https://github.com/google/verible/pull/131

Copybara import of the project:

  - bc70e9910c099b43cf45eae525b7891650628220 Add partitioning around kAssignModifyStatement by Rafal Kapuscik <rkapuscik@antmicro.com>

Closes #131

PiperOrigin-RevId: 290172021
closed 2020-01-17 by hzeller