chipsalliance/verible #95: [line break/wrap] Erroneous line break in if/else without begin/else blocks

msfschaffner, 2019-12-16:
is:
```systemverilog
    if (enable) data |= interrupts;
        else data &= ~interrupts;
```
should:
```systemverilog
    if (enable) data |= interrupts;
    else data &= ~interrupts;
```

msfschaffner, 2019-12-16:
b/146353167

fangism, 2019-12-16:
Confirmed:

Complete test case formats to:

```systemverilog
task t;
    if (enable) data |= interrupts;
        else data &= ~interrupts;
endtask
```

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/formatting/b146353167.sv
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([task t ;]) }
  }
  { (>>[<auto>]) @{1}
    { (>>[<auto>]) @{1,0}
      { (>>[if ( enable ) data |= interrupts ; else data &= ~ interrupts ;]) }
    }
  }
  { ([endtask]) }
}
Halting for diagnostic operation.
```

The `else` clause should start its own partition when it doesn't immediately follow an `end` keyword.

ldalek-antmicro, 2020-01-03:
This issue can be solved by testing that `else` keyword is not preceded by `end` keyword:
https://github.com/google/verible/pull/115/commits/74849e29f900f5dadf83d7f0f15697de2736eeb7#diff-7872d97b05fbc6f2a64cf14b8ae74d75R918

Result:
``` 
if (en1) data |= interrupts;
else data &= ~interrupts;
```

Or by adding `kAssignModifyStatement` in https://github.com/google/verible/blob/master/verilog/formatting/tree_unwrapper.cc#L649

Result:
```
if (en1)
  data |= interrupts;
else
  data &= ~interrupts;
```

I'm not quite sure which one is better. @fangism ?

fangism, 2020-01-03:
My reading of our style guide, (which is also reflected here: https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#begin--end in the `always_ff` example) suggests that if one wraps a single statement block (whether or not it follows `if` or `always`) that it must be enclosed in `begin` and `end`, spanning 3+ lines.  (Personally, both versions look acceptable to me, but this is not my call to make.)  However, the formatter is not in any position at this time to inject `begin` `end` tokens into the output stream (or verify rewrite equivalence in the presence of lexical differences).  There will be occurrences of `if (long_condition) long_statement;`, and I suspect the best we can do now is follow the wrapping rule (+4 spaces) when wrapping the statement.  I think the most practical short-term solution would be to compact-if-it-fits-on-one-line.

@snsokolov @msfschaffner Would you care to comment?

msfschaffner, 2020-01-03:
I think the solution you suggest is fine for the time being. The variant without a `begin ... end` block should't be used with long single statements anyway, so I suspect that there will not be many instances where a line break is actually needed. One possibility would be to throw a warning if such a long statement is encountered...

ldalek-antmicro, 2020-01-09:
Hey @fangism,
Thanks for comment https://github.com/google/verible/pull/115#pullrequestreview-338247233. It was really helpful.

It looks like this issue can be solved by PR #115 and by implementing solution that you described in review (which also fixes #103).

What do you think about closing this issue by PR #115 and proceed with https://github.com/google/verible/pull/115#pullrequestreview-338247233 in #103 ?

fangism, 2020-01-09:
> Hey @fangism,
> Thanks for comment [#115 (review)](https://github.com/google/verible/pull/115#pullrequestreview-338247233). It was really helpful.
> 
> It looks like this issue can be solved by PR #115 and by implementing solution that you described in review (which also fixes #103).
> 
> What do you think about closing this issue by PR #115 and proceed with [#115 (review)](https://github.com/google/verible/pull/115#pullrequestreview-338247233) in #103 ?

Sounds good.

ldalek-antmicro, 2020-01-17:
It looks like this commit https://github.com/google/verible/commit/e2e28a86b2300570fc88e8a095779235e1d8b018 fixes this issue:
<pre>
function foo();
  if (enable) data |= interrupts;
  else data &= ~interrupts;
endfunction
</pre>

token partition tree:
<pre>
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([function foo ( ) ;]) }
  }
  { (>>[<auto>]) @{1}
    { (>>[<auto>]) @{1,0}
      { (>>[<auto>]) @{1,0,0}
        { (>>[if ( enable )]) }
        { (>>>>[<auto>]) @{1,0,0,1}
          { (>>>>[data |= interrupts ;]) }
        }
      }
      { (>>[<auto>]) @{1,0,1}
        { (>>[else]) }
        { (>>>>[<auto>]) @{1,0,1,1}
          { (>>>>[data &= ~ interrupts ;]) }
        }
      }
    }
  }
  { ([endfunction]) }
}
Halting for diagnostic operation.
</pre>

Output:
<pre>
function foo();
  if (enable) data |= interrupts;
  else data &= ~interrupts;
endfunction
</pre>

I guess we can close this issue, @fangism ?

fangism, 2020-01-17:
Thanks for checking back.  

The only bit I have to think about is:

```
        { (>>[if ( enable )]) }
        { (>>>>[]) @{1,0,0,1}   // 3 or 4 > levels if wrapped?
          { (>>>>[data |= interrupts ;]) }
        }
```

the style guide asks for begin/end on single statements that wrap to the next line, and begin/end blocks at one level of indentation.  But that is an easy matter to adjust (potential separate issue), so I'm happy to close this.

closed 2020-01-17 by fangism