chipsalliance/verible #96: [line breaks, spaces] Covergroup/function formatting

msfschaffner, 2019-12-16:
is:
```systemverilog
covergroup intr_test_cg(
    uint num_interrupts
) with function sample (
    uint intr, bit intr_test, bit intr_en, bit intr_state
);
```

this should follow a function/task format, e.g.:

```systemverilog
covergroup intr_test_cg(uint num_interrupts) with function sample(uint intr,
                                                                  bit intr_test,
                                                                  bit intr_en,
                                                                  bit intr_state);
```
or 
```systemverilog
covergroup intr_test_cg(uint num_interrupts) with function sample(uint intr,
    bit intr_test, bit intr_en, bit intr_state);
```

Note also the spaces before opening `(`

msfschaffner, 2019-12-16:
b/146354943

fangism, 2019-12-16:
Confirming behavior, but there are multiple issues here to be broken down, and I'm not sure at this time what the right behavior even is (certainly not explicitly covered in internal style guide yet).  

Whole test case:

```systemverilog
covergroup intr_test_cg(
    uint num_interrupts
) with function sample (
    uint intr, bit intr_test, bit intr_en, bit intr_state
);
endgroup
```

Partition tree dump:

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/formatting/b146354943.sv
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([covergroup intr_test_cg (]) }
    { (>>>>[<auto>]) @{0,1}
      { (>>>>[uint num_interrupts]) }
    }
    { ([) with function sample (]) }
    { (>>>>[<auto>]) @{0,3}
      { (>>>>[uint intr ,]) }
      { (>>>>[bit intr_test ,]) }
      { (>>>>[bit intr_en ,]) }
      { (>>>>[bit intr_state]) }
    }
    { ([) ;]) }
  }
  { ([endgroup]) }
}
```