chipsalliance/verible #97: [line breaks] DPI-C function formatting

msfschaffner, 2019-12-16:
is:
```systemverilog
  import "DPI-C" function chandle spidpi_create(
      input string name, input int mode, input int loglevel
  );
```
should:
```systemverilog
  import "DPI-C" function chandle spidpi_create(input string name, input int mode, input int loglevel);
```
if it does not fit onto a single line, we prefer the following formats:
```systemverilog
  import "DPI-C" function chandle spidpi_create(
      input string name, 
      input int mode,
      input int loglevel
  );
```
```systemverilog
  import "DPI-C" function chandle spidpi_create(input string name, 
                                                input int mode,
                                                input int loglevel);
```

msfschaffner, 2019-12-16:
b/146354843

fangism, 2019-12-16:
Confirming with complete test case:

```systemverilog
module m;
  import "DPI-C" function chandle spidpi_create( 
      input string name, input int mode, input int loglevel
  );
endmodule
```

Indeed, that import declaration exceeds the 100-column limit.

Current partitioning:

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/formatting/b146354843.sv
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([module m ;]) }
  }
  { (>>[<auto>]) @{1}
    { (>>[import "DPI-C" function chandle spidpi_create (]) }
    { (>>>>>>[<auto>]) @{1,1}
      { (>>>>>>[input string name ,]) }
      { (>>>>>>[input int mode ,]) }
      { (>>>>>>[input int loglevel]) }
    }
    { (>>[) ;]) }
  }
  { ([endmodule]) }
}
```

Closely related to #29.

Agree that either one of your "preferred formats" looks better.

commit 9e55307a (2020-03-05): PR #223: Fix DPI-C function formatting
Fixes #97

GitHub PR https://github.com/google/verible/pull/223

Copybara import of the project:

  - f6c691a76d9ce47dc7cfe967fc383f9e3c2312db Reshape kDPIImportItem partitions and apply kAppendFittin... by Lukasz Dalek <ldalek@antmicro.com>

Closes #223

PiperOrigin-RevId: 299139688
closed 2020-03-05 by hzeller