chipsalliance/verible #98: [line breaks] One-line case statements without begin/end blocks

msfschaffner, 2019-12-16:
is:
```systemverilog
case (compare_op)
  CompareOpEq: if (read_data == exp_data)
    break;
  CompareOpCaseEq: if (read_data === exp_data)
    break;
  CompareOpNe: if (read_data != exp_data)
    break;
  CompareOpCaseNe: if (read_data !== exp_data)
    break;
  CompareOpGt: if (read_data >   exp_data)
    break;
  CompareOpGe: if (read_data >= exp_data)
    break;
  CompareOpLt: if (read_data <   exp_data)
    break;
  CompareOpLe: if (read_data <= exp_data)
    break;
  default: begin
    `uvm_fatal(ptr.get_full_name(), $sformatf("invalid operator:%0s", compare_op))
  end
endcase
```
should:
```systemverilog
case (compare_op)
  CompareOpEq:     if (read_data ==  exp_data) break;
  CompareOpCaseEq: if (read_data === exp_data) break;
  CompareOpNe:     if (read_data !=  exp_data) break;
  CompareOpCaseNe: if (read_data !== exp_data) break;
  CompareOpGt:     if (read_data >   exp_data) break;
  CompareOpGe:     if (read_data >=  exp_data) break;
  CompareOpLt:     if (read_data <   exp_data) break;
  CompareOpLe:     if (read_data <=  exp_data) break;
  default: begin
    `uvm_fatal(ptr.get_full_name(), $sformatf("invalid operator:%0s", compare_op))
  end
endcase
```

msfschaffner, 2019-12-16:
b/146354852

msfschaffner, 2019-12-16:
another related example:

is:
```systemverilog
if (cfg.if_mode == Host)
  driver = HOST_DRIVER_T::type_id::create("driver", this);
else
  driver = DEVICE_DRIVER_T::type_id::create("driver", this);
```
should:
```systemverilog
if (cfg.if_mode == Host)  driver = HOST_DRIVER_T::type_id::create("driver", this);
else                      driver = DEVICE_DRIVER_T::type_id::create("driver", this);
```

fangism, 2019-12-16:
Focusing on original example only (case statement) in this issue.  (File the if-else example in a separate issue, please.)

I suspect this can be attributed to:

https://github.com/google/verible/blob/78170927a2f881a6189a71a871b62701eb6ddfa0/verilog/formatting/tree_unwrapper.cc#L566

A case-item should be allowed to fit on a single-line.

commit 43aa7aea (2020-01-13): PR #123: Allow case items to fit on one line
Fixes #98.
This PR does not address tabular alignment.

GitHub PR https://github.com/google/verible/pull/123

Copybara import of the project:

  - a5f4b3adf07e1623baa1fb91f27495c01c0a494e Allow case items to fit on one line by Rafal Kapuscik <rkapuscik@antmicro.com>

Closes #123

PiperOrigin-RevId: 289504098
closed 2020-01-13 by hzeller