chipsalliance/verible #99: [line breaks] Missing line break after `join_any`

msfschaffner, 2019-12-16:
is:

```systemverilog
join_any disable fork;
```

should:
```systemverilog
join_any
disable fork;
```

msfschaffner, 2019-12-16:
b/146355401

fangism, 2019-12-16:
Confirmed.

Complete test case:

```systemverilog
task t;
fork
join_any
disable fork;
endtask
```

```
$ verilog_format --show_token_partition_tree ~/testing/verilog/formatting/b146355401.sv
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([task t ;]) }
  }
  { (>>[<auto>]) @{1}
    { (>>[fork]) }
    { (>>[join_any disable fork ;]) }
  }
  { ([endtask]) }
}
```

`disable fork;` just needs to start its own partition.

```
$ verilog_syntax --printtree  ~/testing/verilog/formatting/b146355401.sv
Parse Tree: 
Node (tag: kDescriptionList) {
  Node (tag: kTaskDeclaration) {
    Node (tag: kTaskHeader) {
      (#"task" @0-4: "task")
      Node (tag: kUnqualifiedId) {
        (#SymbolIdentifier @5-6: "t")
      }
      (#';' @6-7: ";")
    }
    Node (tag: kStatementList) {
      Node (tag: kParBlock) {
        (#"fork" @8-12: "fork")
        Node (tag: kBlockItemStatementList) {
        }
        (#"join_any" @13-21: "join_any")
      }
      Node (tag: kDisableStatement) {
        (#"disable" @22-29: "disable")
        (#"fork" @30-34: "fork")
        (#';' @34-35: ";")
      }
    }
    (#"endtask" @36-43: "endtask")
  }
}
```

and that would be `kDisableStatement` in tree_unwrapper.cc.

commit 92186b65 (2020-01-13): PR #122: Add partitioning around kDisableStatement
Fixes #99.

GitHub PR https://github.com/google/verible/pull/122

Copybara import of the project:

  - 0e3f54f1fcd5b5c8e1914b48de421ec9de13a69c Add partitioning around kDisableStatement by Rafal Kapuscik <rkapuscik@antmicro.com>

Closes #122

PiperOrigin-RevId: 289516109
closed 2020-01-13 by hzeller