chipsalliance/verible #102: [style-lint] Syntax error within SV macros

msfschaffner, 2019-12-17:
The verible style linter tries to analyze SV syntax within macros, which can sometimes fail due to missing context. E.g., when run on this file:

https://github.com/google/riscv-dv/blob/master/src/riscv_instr_cover_group.sv

I get many errors of the sort:
```
./hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv:220:18: syntax error, rejected ":" (go/verible#syntax-error).
./hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv:224:18: syntax error, rejected ":" (go/verible#syntax-error).
./hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv:228:18: syntax error, rejected ":" (go/verible#syntax-error).
./hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv:239:18: syntax error, rejected ":" (go/verible#syntax-error).
./hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv:243:18: syntax error, rejected ":" (go/verible#syntax-error).
./hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv:247:18: syntax error, rejected ":" (go/verible#syntax-error).
```

msfschaffner, 2019-12-17:
b/146383763

fangism, 2019-12-17:
You are burying unbalanced begin/end (covergroup/endgroup) constructs inside a macro definition, that's asking for trouble.  The verible syntax checker runs unpreprocessed, and doesn't even attempt to expand locally defined macros.

What you're essentially asking for in this issue is:
1) track local macro definitions (already done, but not used)
2) expand locally defined macros between lexing and parsing (not done)
3) interpret expanded syntax tree (where expanded nodes have performed some sort of text synthesis and substitution)

msfschaffner, 2019-12-17:
albeit 1-3. would be nice to have, I think it would for the time being even be fine to leverage 1. to stop linting anything that is inside the body of the macro definition, WDYT?

fangism, 2019-12-17:
More precisely, this isn't even a lint issue; it is being reported by the syntax checker, before any linting is done.

1 alone doesn't beget any user-visible improvement in behavior.

weicaiyang, 2020-01-09:
Found couple similar errors from this file:
https://github.com/lowRISC/opentitan/blob/master/hw/dv/sv/sw_msg_monitor_if/sw_msg_monitor_if.sv

```
./sw_msg_monitor_if/sw_msg_monitor_if.sv:368:7: syntax error, rejected "`NARGS_CASE" (go/verible#syntax-error).
./sw_msg_monitor_if/sw_msg_monitor_if.sv:401:5: syntax error, rejected "endcase" (go/verible#syntax-error).

```

msfschaffner, 2020-01-10:
Can we work around this error with an inline waiver, or similar?

fangism, 2020-01-10:
Syntax errors cannot be waived.  Only lint rule violations can be waived.

fangism, 2020-01-10:
b/37489120
b/34791117
b/111544845

fangism, 2020-01-10:
> What you're essentially asking for in this issue is:
> 
> 1. track local macro definitions (already done, but not used)
> 2. expand locally defined macros between lexing and parsing (not done)

I'm not certain at this time that even if this were implemented that it would always be the right thing to do to analyze the expanded form.  If a lint rule is about macro usage itself, the right thing may be to leave it un-expanded.  I think we need a deeper study about these cases before acting on it.

> 3. interpret expanded syntax tree (where expanded nodes have performed some sort of text synthesis and substitution)

Note that this interpretation is only applicable to analysis, certainly not formatting.  However, since the formatter depends on the parser and resulting syntax tree in its current architecture, any syntax error (false positive or not) will preclude formatting.  If expansion of a particular macro helps the parser continue, then we may need some sort of reverse transformation, where the analyzed expanded tree is viewed/operated-on in its *un-expanded* form for formatting purposes, e.g. replace a case-item subtree with a macro-call node.