chipsalliance/verible #103: [line breaks] One-line if/else statements without begin/end blocks

msfschaffner, 2019-12-17:
splitting out from #98.
 
is:
```systemverilog
if (cfg.if_mode == Host)
  driver = HOST_DRIVER_T::type_id::create("driver", this);
else
  driver = DEVICE_DRIVER_T::type_id::create("driver", this);
```
should:
```systemverilog
if (cfg.if_mode == Host)  driver = HOST_DRIVER_T::type_id::create("driver", this);
else                      driver = DEVICE_DRIVER_T::type_id::create("driver", this);
```

msfschaffner, 2019-12-17:
b/146384260

ldalek-antmicro, 2020-01-03:
Looks like related to #95

fangism, 2020-01-03:
I think what @msfschaffner is emphasizing here is that single-statement clauses should be compacted onto the same line as `if ...` and `else` if they fit within the column limit.  #95 is about missing handling of `else`.

This may be helped by some of the comments about introducing CST nodes for `kIfClause` and `kElseClause` described here:

https://github.com/google/verible/pull/115#pullrequestreview-338247233

Introducing a level of partitioning will make it easier to apply the `kFitOnLineElseExpand` policy, and still provide opportunities to manipulate the partition tree.

Lastly, Michael's "should" example demonstrates statement alignment across clauses, which is not required to address at this time.

ldalek-antmicro, 2020-01-15:
I've implemented what you've suggested in https://github.com/google/verible/pull/115#pullrequestreview-338247233 but also I've added
`kIfBody` and `kElseBody` for handling proper indentation in If-else statements, e.g.
```
      { (>>[<auto>]) @{16,0,1}
        { (>>[else if ( yy )]) }
        { (>>>>[<auto>]) @{16,0,1,1}
          { (>>>>[return 1 ;]) }
        }
      }
```

commit 0d45afbb (2020-01-16): PR #120: [line breaks] One-line if/else statements without begin/end blocks
Fix #103

GitHub PR https://github.com/google/verible/pull/120

Copybara import of the project:

  - 40ed7080fce57228614a4e3ca2584ec99ce7601c Introduce new conditional statements node tags by Lukasz Dalek <ldalek@antmicro.com>
  - d671b6fbfd75528021c7500c5d4090d0c7ecf2b0 Group kConditionalStatement subnodes by Lukasz Dalek <ldalek@antmicro.com>
  - 859002ed318a2a752564e41fe0b3adc4798b956b Add SpanPrevToken() method to UnwrappedLine class by Lukasz Dalek <ldalek@antmicro.com>
  - 311b334e835c1be61425a7ca4fe7295a688e92bc [... cut]
closed 2020-01-16 by hzeller