chipsalliance/verible #104: [line breaks, wraps] Parameterized class declaration

msfschaffner, 2019-12-17:
splitting out from #34

should:
```systemverilog
class cip_base_scoreboard #(type RAL_T = dv_base_reg_block,
                            type CFG_T = cip_base_env_cfg,
                            type COV_T = cip_base_env_cov)
```

is:
```systemverilog
class cip_base_scoreboard #(
    type RAL_T = dv_base_reg_block, type CFG_T = cip_base_env_cfg, type COV_T = cip_base_env_cov
)
```

msfschaffner, 2019-12-17:
b/146384248

fangism, 2019-12-17:
Would the following be acceptable?

```systemverilog
class cip_base_scoreboard #(
    type RAL_T = dv_base_reg_block,
    type CFG_T = cip_base_env_cfg,
    type COV_T = cip_base_env_cov);
endclass
```

fangism, 2019-12-17:
I meant:

```systemverilog
class cip_base_scoreboard #(
    type RAL_T = dv_base_reg_block,
    type CFG_T = cip_base_env_cfg,
    type COV_T = cip_base_env_cov
);
endclass
```

msfschaffner, 2019-12-17:
Yes I think that would be ok as well.

commit b1628f95 (2019-12-21): PR #112: Fix class parameters indentation
Fix #104

GitHub PR https://github.com/google/verible/pull/112

Copybara import of the project:

  - d7e480ef66becd443e45aeb93a822d2501898457 Fix class parameters indentation by Lukasz Dalek <ldalek@antmicro.com>

Closes #112

PiperOrigin-RevId: 286658908
closed 2019-12-21 by hzeller