chipsalliance/verible #105: Support for separate waiver files

msfschaffner, 2019-12-17:
It would be convenient if the Verible style linter supported separate linter files. For instance, in OpenTitan we currently already leverage two linting solutions (including Verilator lint). Adding yet another linter and allowing in-line waivers would lead to unacceptable code clutter.

msfschaffner, 2019-12-17:
b/146448516

msfschaffner, 2019-12-17:
In terms of waiver format, I believe a similar format as used in Verilator would be sufficient (i.e., full regexp matching on the error message output is not strictly required).

E.g.:
```
// Bits of signal are not used: boot_addr_i[7:0]
// Boot address is 256B aligned, cleaner to pass all bits in
lint_off -msg UNUSED -file "*/rtl/ibex_if_stage.sv" -lines 40
```

See for example here for a verilator lint waiver file:
https://github.com/lowRISC/ibex/blob/master/lint/verilator_waiver.vlt

snsokolov, 2020-02-01:
Just to clarify, this is asking for a single global waiver config file and not for per-file waivers (I advice against per-file waivers)?

Internally, we wrap the Linter with the hardcoded support of source sub-tree based per-rule disables. But I think a global config file (or just a waiver config file) proposed in this bug would be a better alternative to this and work externally and internally.

fangism, 2020-02-01:
My understanding was that this request was about supporting storing waivers (including some aimed at specific lines) in a completely separate file.  This avoid polluting source files with waivers, as some prefer.

If we had https://github.com/google/verible/issues/15 resolved, we could store flags in a flag file as a poor man's configuration.

msfschaffner, 2020-02-10:
@snsokolov actually this was asking for support for per-file waivers that can be stored alongside the source files. 

Let me explain why we are interested in such a feature:

1) inline waivers are fine of you only have one or two linting tools, but since OT is opensource, we are going to have several lint tools - both commercial and non-commercial. for instance, we already have verilator lint, ascent lint, and are soon going to add spyglass and verilator linting solutions as well. inline waivers do not scale with this, and they have a tendency to clutter the code base.

2) we would like to be able to group the waivers with the modules that they belong to, since depending on the system configuration, certain modules may be included in the build, but others may not. our build flow only includes the waivers that are required for a specific build.

3) we have multiple designers in different time zones from different organizations which may not be familiar with the complete code base, but only with their IP they are working on. the modular approach from 2. helps to locate and adjust waivers quickly.

You can find examples in the IP directories in OpenTitan. For instance, the following folders contain the lint waivers for the AES and HMAC modules:

https://github.com/lowRISC/opentitan/tree/master/hw/ip/aes/lint
https://github.com/lowRISC/opentitan/tree/master/hw/ip/hmac/lint

I think I agree with your point that we should not have per-file lint configurations, but if I understand this correctly, the lint waivers I am talking about do not represent the rule configurations that would disable the rules on a per-file basis, but merely act as a message filter such that the warnings and errors do not show in the final report (i.e., it is a sort of report post-processing). For instance, we only use one global AscentLint policy file that configures the rules for OpenTitan.

fangism, 2020-02-10:
I'm open to hearing proposals about how you think such a waiver file should be structured: {rule, file(s), ranges, patterns, etc...}.
What I'm thinking is that once violations are collected, they are then matched up against waiver sets to be filtered out.
External waiver sets could be loaded with a new flag: `verilog_lint --waiver_file=FILE` (where file could apply to a whole project)

msfschaffner, 2020-02-11:
I would stay pragmatic and follow the structure that other tools use in this space. 

For instance, Verilator (which is also open source and which also has a built in linter) supports the following waiver formats (boldly copying from https://www.veripool.org/ftp/verilator_doc.pdf):
```
lint_on [-rule <message>] [-file "<filename>" [-lines <line> [ - <line>]]]
lint_off [-rule <message>] [-file "<filename>" [-lines <line> [ - <line>]]]
lint_off [-rule <message>] [-file "<filename>"] [-match "<string>"]
Enable/disables the specified lint warning, in the specified filename (or wildcard
with '*' or '?', or all files if omitted) and range of line numbers (or all lines if
omitted).
With lint_off using '*' will override any lint_on directives in the source, i.e.
the warning will still not be printed.
If the -rule is omitted, all lint warnings (see list in -Wno-lint) are enabled/disabled.
This will override all later lint warning enables for the specified region.
If -match is set the linter warnings are matched against this (wildcard) string
and are waived in case they match iff rule and file (with wildcard) also match.
In previous versions -rule was named -msg. The latter is deprecated, but still
works with a deprecation info, it may be removed in future versions.
```
AscentLint from RealIntent follows a similar format, where you can do something like:
```
waive -rules <name> [-location <filename>] [-regexp <regular expression>] [-comment <comment>]
waive -rules <name> [-location <filename>] [-msg <message string>] [-comment <comment>]
```
Now I should add that the line-based format in the first two waiver formats from Verilator above can be quite unstable - especially considering that we eventually plan to autoformat the code in presubmits. So it would be best if the waiver format followed the "location + regexp" approach. 

It should also be noted that each lint run will likely have multiple waiver source files (one per IP), but since the location of the messages to be waived can be encoded in the waiver statements, it is permissible to treat all these waiver files as as one large concatenated file.

snsokolov, 2020-02-12:
@msfschaffner I got your point. Curious, do you have examples of such per-file waiver files, maybe I'll change my mind.

From my experience maintaining the Style(!) Linter for multiple parallel projects, I still think that using per-file waivers is methodologically not right, reasons:

1. "Project" (and not particular block/file owner) decides that they going to use a linter tool
2. "Project" decides which rules they want to apply to all files or group of files
3. It is expected that any developer working on the "Project" and violating the Lint rule either to fix it or properly document the violation waiver inline in the code, so other people won't be surprised during code review or refactoring.

I think it is reasonable to have some documentation inside the code about intentional violations to project-wide style rules. What do you think?

msfschaffner, 2020-02-12:
You can find lint waivers in each IP directory in the OpenTitan repo under `<ip name>/lint`:
https://github.com/lowRISC/opentitan/tree/master/hw/ip/

I agree with your points, and we are following 1.-3. as you described them - the only difference is where we actually place those waivers. I do agree that inline waivers have their benefit in terms of documentation value, and I would say that in a different context I would likely also prefer that approach. 

But I also think that there is value in supporting separate waiver files since other projects with different environments may have valid reasons for not placing these waivers directly into the source code even though inline waivers would be methodologically "cleaner". 

Within OpenTitan, some of the main reasons why we place the waivers into separate files are:

1) We have several different linting solutions in place, and allowing inline waivers for all of them does in our view not scale. at the moment, we are already working with three tools (Verilator, AscentLint, Verible), and more will likely be added over the course of this year for demonstration purposes. Some of these linting solutions may also carry style elements such as long lines or naming of code block labels.

2) Files that contain generated code may only partially follow some of the lint rules, depending on who owns the generator scripts and how strictly these can be controlled.

3) The project is opensource, and as such may have several consumers of the code which may have an entirely different tooling environment. Inlining the waivers basically imposes our solutions onto them, which means that they would have to always apply a patch on top of upstream.

Don't get me wrong - I do not want to force this feature onto you :), hence I mentioned at the beginning of this thread that the feature would be *convenient*. There are ways to make it work even without support for separate waiver files, for example by placing all style-related Verible waivers in line, and other waivers from other tools into separate files. 

I do agree that separate waivers may feel like a compromise and may not be as methodologically clean. But on the other hand, separate waiver files have valid use cases and by supporting this feature the tool would become more user friendly.

I should also mention that most tools I have seen so far (commercial and opensource like Verilator) support both lint waiver formats.

imphil, 2020-07-02:
This was fixed in #245 and this issue can be closed.

closed 2020-07-02 by mithro