chipsalliance/verible #107: always-ff-non-blocking rule wrongly flags for-loop-step assignment

fangism, 2019-12-17:
On this file:

```systemverilog
module m;
always_ff @(posedge clk) begin
 for(int i=0; i<Num_irq; i=i+1) begin    // linter flags the "i=i+1" assignment
 end
 end
 endmodule
```

```
$ verilog_lint --ruleset=none --rules=always-ff-non-blocking ~/testing/verilog/always_ff_for.sv
~/testing/verilog/always_ff_for.sv:3:27: Use only non-blocking assignments inside 'always_ff' sequential blocks. go/verilog-style#sequential-logic [always-ff-non-blocking]
```

Assignments inside for-loop-headers should be exempt from this check.

https://github.com/google/verible/blob/73936a870a7948663cc1009cd8b36a40f3ebe69d/verilog/analysis/checkers/always_ff_non_blocking_rule.cc#L56

In this function, one could examine the context, to see whether or not it contains `kForStepList` or `kLoopHeader`.

fangism, 2019-12-17:
b/146457597

fangism, 2019-12-17:
@ldalek-antmicro would you like to take this one, since you worked on the original rule implementation?

ldalek-antmicro, 2019-12-18:
@fangism I've prepared PR #109. Can you look at it?

commit 7c83bd80 (2019-12-18): PR #109: lint: Ignore blocking statement in loop header
Fix #107

GitHub PR https://github.com/google/verible/pull/109

Copybara import of the project:

  - b03aed92d9abb601ed3624984ff93d6975a694fc lint: Ignore blocking statement in loop header by Lukasz Dalek <ldalek@antmicro.com>

Closes #109

PiperOrigin-RevId: 286227766
closed 2019-12-18 by hzeller