chipsalliance/verible #127: Detect space between case item and colon, function/task/macro call and open parenthesis

ldalek-antmicro, 2020-01-15:
As described in https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#basic-style-elements 6th bullet

fangism, 2020-01-15:
True.  As a general policy, we leave all formatting rules out of the style linter if it is expected that the formatter would fix the issue, reason being that users might get aggravated at the number of spacing violations reported :) .  In fact, the intent is to be able to use the formatter in a lint-like mode to describe style violations, because it already has all the information needed to correct spaces and would just need to be extended to report detailed changes (possibly citing reasons).  Imagine an annotated diff between input and formatted output.

fangism, 2020-01-15:
This value wrapper

https://github.com/google/verible/blob/master/common/util/with_reason.h

kind of carries the intent to relay reasons to the debugger and end-user.

if (original_space != adjusted_space) print reason;

ldalek-antmicro, 2020-01-16:
Okay. Because Formatter is able to fix this cases:
<pre>
initial begin
  case (x)
    default : break;
    1'b0    : break;
  endcase
end

module m;
function foo ( ) ; endfunction
initial foo ( ) ;
endmodule

module m;
task foo ( ) ; endtask
initial foo ( ) ;
endmodule

module m;
initial begin `foo ( ) ; end
endmodule
</pre>

as

<pre>
initial begin
  case (x)
    default: break;
    1'b0: break;
  endcase
end

module m;
  function foo();
  endfunction
  initial foo();
endmodule

module m;
  task foo();
  endtask
  initial foo();
endmodule

module m;
  initial begin
    `foo();
  end
endmodule
</pre>

this issue can be closed?

fangism, 2020-01-16:
Yes, those cases look like they're already working in the formatter, and I believe we have representative test cases that cover those, so I'll close.

closed 2020-01-16 by fangism