chipsalliance/verible #128: Nested struct typedefs

msfschaffner, 2020-01-15:
I get the following error when running style lint on nested struct definitions:
```
struct definitions always should be named using typedef. [Style: typedef-structs-unions] [typedef-structs-unions]
```
example snippet:
```systemverilog
  typedef struct packed {
    struct packed {
      logic [15:0] q;
      logic        qe;
    } data;
    struct packed {
      logic [15:0] q;
      logic        qe;
    } mask;
  } gpio_reg2hw_masked_out_upper_reg_t;
```
the keyword `typedef` is not allowed in front of the nested structs, since they basically do not represent definitions themselves, but belong to the type being defined.

msfschaffner, 2020-01-15:
b/147759348

fangism, 2020-01-15:
There was some internal discussion that resolved that even these internally nested structs should be named with a separate `typedef` outside of the outer struct.

msfschaffner, 2020-01-16:
Got it, thanks. 

In our case this appears in generated packages, where it was convenient to nest some of the structs. Here's an example:

https://github.com/lowRISC/opentitan/blob/ee57d1aec641cd4f72b7853c5ad841edfd2c1159/hw/ip/gpio/rtl/gpio_reg_pkg.sv#L57-L66

fangism, 2020-02-02:
Once we have configurable lint rules from https://github.com/google/verible/issues/128, it would be easier to fine-tune for classes of exceptions like this.

mithro, 2020-02-02:
I think the error message is misleading in this case as doing the "dump" solution causes a invalid System Verilog. Instead the error should be about nested structs inside a definition.

mithro, 2020-02-02:
IE https://github.com/google/verible/issues/170

> Look at this code:
> 
> ```verilog
>   typedef struct packed {
>     struct packed {
>       logic        q;
>     } classa;
>   } alert_handler_reg2hw_intr_state_reg_t;
> ```
> 
> https://github.com/lowRISC/opentitan/blob/ebcabe267a093b2eba868f4ca9128826fefff74b/hw/top_earlgrey/ip/alert_handler/rtl/autogen/alert_handler_reg_pkg.sv#L26-L39
> 
> Note that adding a nested `typedef` to the second struct is not valid Verilog.
> 
> Running verilog_lint on this code gives us:
> 
> ```
> alert_handler_reg_pkg.sv:27:5: struct definitions always should be named using typedef. [Style: typedef-structs-unions] [typedef-structs-unions]
> ```
> 
> Following this advice leads to invalid code (the parser knows that and will error appropriately if there's a nested typedef).

fangism, 2020-02-02:
True, we could improve the error message today by detecting the nested case (easy) and recommending defining a `typedef` for an inner struct outside of the outermost struct.

msfschaffner, 2020-02-15:
Could we make this rule configurable and add another mode that is more permissible? I.e. a mode that allows nested struct definitions, and only checks for `typedef` at the outermost level?

fangism, 2020-02-15:
Yes.  Blocked by #133

commit a9d1b641 (2020-03-24): Lint: Make acceptance of anonymous nested structs configurable.

This adds 'allow_anonymous_nested' configuration to the 'typedef-structs-unions'
rule to allow not to require a typedef for inner structs.

In the course of that, implement SyntaxTreeContext::IsInsideStartingFrom()
which starts looking backward the stack starting with an offset.

(As a side effect, make IsInside() to start looking from top of stack
instead of bottom, which is conceptually more aligned of what we expect,
even though it makes only a difference in this case (offset 0))

Fixes #128

PiperOrigin-RevId: 302738764
closed 2020-03-24 by hzeller