chipsalliance/verible #133: Support parameterizable lint rules

fangism, 2020-01-17:
Try as we may, it is impossible to get all users to agree on style.  To reach more users with the style linter, we must accommodate different tastes.

For some lint rules, it makes sense to be able to parameterize:
Naming style rules: encode various naming styles with enums for choices
Quantitative rules: line-length, max-blank-lines

Today, command-lines flags (and `--flagfile`) are the only way to enable/disable selected rules at run-time.
`--rules=RULENAME` enables a rules and `--rules=-RULENAME` disables a rule, very binary.  Multiple on/off elections are comma-separated.

The proposal here is to enable every rule to take a single string, parse it however it sees fit, and operate accordingly.

Command-line specification:
`--rules=RULENAME[=text],...`
`text` itself takes any form, may support separators, may even have structure.

Parameterized rules should be implemented with defaults, so that when `=text` is omitted, it still functions.  This is different from `RULENAME=` (blank), which intentionally passes an empty string to the parameter parser (which may differ from the default configuration).

Parameterless rules unchanged.
`--rules=-RULENAME` will still disable a rule entirely.
same with `--rules=-RULENAME=text...` (so that it is easy to switch with a single `-` edit)

Rule classes' interface should be updated to support a single parameter string in the constructor, which is passed via the factory function that instantiates rule objects.

There should be proper error-handling of parameters, including when parameters are unexpected, parse-incorrectly, are invalid enums, etc.   Since we do not use exceptions, there will need to be some status that indicates whether a rule object was properly constructed.  Errors should produce `--help`-like messages that get relayed back to the user, and cause a non-zero exit status.

With proposed enhancements, integration tests should be able to support "rule-name=VALUE" in the BUILD file (forwarded CL arguments).

Unit-testing framework for individual lint rules will need to be able to accept parameterized rules.

Will need additional parameter validation, like enum_flags does.  The enum_flags library will probably come in handy:
https://github.com/google/verible/blob/master/common/util/enum_flags.h

Parsing and Unparsing of parameters/strings should be exact reverses.

Diagnostics on found lint rule violations will need to report parameter (along with rule name) if applicable.  e.g. `[line-length=100]` `[blah-name-style=snake]`.

Documentation (generated) should reflect parameter usage for each rule.

fangism, 2020-01-17:
b/144328056

hzeller, 2020-02-25:
This is now implemented to the extend that the infrastructure is there to configure rules via the proposed flag `<rule>=<config>,...` syntax as described above. [Generated documentation](https://google.github.io/verible/lint.html#line-length) contains accepted parameters. First implementation for `line-length` lint rule is done.

Leaving this open for now until
   * We have a common configuration parser/validator to simplify writing new configuration rules.
   * a unit-test framework that allows to easily test variants.
   * Also a framework in which most of the documentation boilerplate is auto-generated (e.g. with the rule announcing all the parameters it accepts, their type and range: automatic validation and documentation generation)

hzeller, 2020-04-07:
I'll close this now - there is a framework to configure lint rules, and it is used by a few of them:

https://google.github.io/verible/lint.html#explicit-parameter-storage-type
https://google.github.io/verible/lint.html#line-length
https://google.github.io/verible/lint.html#parameter-name-style
https://google.github.io/verible/lint.html#typedef-structs-unions

documentation is provided, but it would be good if this was more auto-generated. Working on that separately.

closed 2020-04-07 by hzeller