chipsalliance/verible #134: Begin keyword must be on the same line as the preceding keyword and end the line

ldalek-antmicro, 2020-01-17:
Linter shoud issue warning when begin word is not at the same line as proceeding word nor at end the line.

Correct example:
<pre>
module test_always_ff;
always_ff @(posedge clk) begin
  q <= d;
end
endmodule
</pre>

Incorrect:
<pre>
module test_always_ff;
always_ff @(posedge clk)
begin
  q <= d;
end
endmodule
</pre>

fangism, 2020-01-17:
Since this is a white-space only difference, this is something the formatter can and should fix, instead of just informing the user.  Here's the current behavior I see:

```systemverilog
module test_always_ff;
always_ff @(posedge clk)
begin
  q <= d;
end
endmodule
```

formats to:

```systemverilog
module test_always_ff;
  always_ff @(posedge clk)
begin
    q <= d;
  end
endmodule
```

but should be

```systemverilog
module test_always_ff;
  always_ff @(posedge clk) begin
    q <= d;
  end
endmodule
```

The `begin` should merged into the partition with `always_ff`, similar to `if (...) begin`.  I'd recommend we fix the formatter without a separate lint rule.

We are also using the formatter in a way that informs the user of all changes that would be made, which in a way uses the formatter as a white-space linter.  It just `diff -u`'s the input and output, and asks the user to reformat if she agrees with the suggested result.

fangism, 2020-01-17:
More detail on this case:

```
Full token partition tree:
{ ([<auto>]) @{}
  { ([<auto>]) @{0}
    { ([module test_always_ff ;]) }
  }
  { (>>[<auto>]) @{1}
    { (>>[always_ff @ ( posedge clk ) begin]) }     // this is correct partitioning
    { (>>>>[<auto>]) @{1,1}
      { (>>>>[<auto>]) @{1,1,0}
        { (>>>>[q <= d ;]) }
      }
    }
    { (>>[end]) }
  }
  { ([endmodule]) }
}
```

The reason it looks the way it does is because of inter-token space handling:

```
verilog_format -v=4 --alsologtostderr ~/testing/verilog/begin-spacing.sv
...
I0117 12:00:09.824245  166114 token_annotator.cc:115] Spacing between ')' and "begin"
I0117 12:00:09.824249  166114 token_annotator.cc:412] spaces: -1, reason: Default: spacing not explicitly handled
```

fangism, 2020-01-21:
b/148097888

commit dbb68e1a (2020-01-21): PR #138: Begin keyword must be on the same line as preceeding ')' token
Fixes #134

GitHub PR https://github.com/google/verible/pull/138

Copybara import of the project:

  - 16b32b7c2a4c5de034645f8fc064cfdd82b8f9a4 Begin keyword must be on the same line as preceeding ')' ... by Lukasz Dalek <ldalek@antmicro.com>

Closes #138

PiperOrigin-RevId: 290810623
closed 2020-01-21 by hzeller