chipsalliance/verible #154: end-else-if-begin should be on same line (partition)

fangism, 2020-01-29:
Input:

```systemverilog
module m;
if (x) begin:z1
assign x=y;
end
else
if (y) begin:z2
assign z=y;
end
endmodule
```

Currently formats to:

```systemverilog
module m;
  if (x) begin : z1
    assign x = y;
  end else
  if (y) begin : z2
    assign z = y;
  end
endmodule
```

but should be:

```systemverilog
module m;
  if (x) begin : z1
    assign x = y;
  end else if (y) begin : z2
    assign z = y;
  end
endmodule
```

With end-else-if-begin on the same line (partition).

fangism, 2020-01-29:
b/148505401

ldalek-antmicro, 2020-02-04:
Hey @fangism,
@hzeller added commit https://github.com/google/verible/commit/ceda448ae637b7665fbd1be6f0672ea172b1f4f0 that fixes this issue:

<pre>
module m;
if (x) begin:z1
assign x=y;
end
else
if (y) begin:z2
assign z=y;
end
endmodule
</pre>

token partition tree:
<pre>
Full token partition tree:
{ ([<auto>]) @{}, policy: always-expand
  { ([<auto>]) @{0}, policy: fit-else-expand
    { ([module m ;]) }
  }
  { (>>[<auto>]) @{1}, policy: always-expand
    { (>>[<auto>]) @{1,0}, policy: fit-else-expand
      { (>>[if ( x ) begin : z1]) }
      { (>>>>[<auto>]) @{1,0,1}, policy: always-expand
        { (>>>>[<auto>]) @{1,0,1,0}, policy: fit-else-expand
          { (>>>>[assign x = y ;]) }
        }
      }
      { (>>[end else if ( y ) begin : z2]) }
      { (>>>>[<auto>]) @{1,0,3}, policy: always-expand
        { (>>>>[<auto>]) @{1,0,3,0}, policy: fit-else-expand
          { (>>>>[assign z = y ;]) }
        }
      }
      { (>>[end]) }
    }
  }
  { ([endmodule]) }
}
</pre>

output:
<pre>
module m;
  if (x) begin : z1
    assign x = y;
  end else if (y) begin : z2
    assign z = y;
  end
endmodule
</pre>

Can you confirm?

ldalek-antmicro, 2020-02-04:
More examples:
----
Input:
<pre>
module zc;
initial begin
if (x) begin
end : l1
else begin
end
end
endmodule
</pre>
Output:
<pre>
module zc;
  initial begin
    if (x) begin
    end : l1
    else begin
    end
  end
endmodule
</pre>
----
Input:
<pre>
module zv;
if (x) begin
end : l1
else begin
end
endmodule
</pre>
Output:
<pre>
module zv;
  if (x) begin
  end : l1
  else begin
  end
endmodule
</pre>
----
Input:
<pre>
module zw;
if (x) begin
end
else begin
end
endmodule
</pre>
Output:
<pre>
module zw;
  if (x) begin
  end else begin
  end
endmodule
</pre>
----
Input:
<pre>
module zx;
if (x) assign z=y;
else assign x=y;
endmodule
</pre>
Output:
<pre>
module zx;
  if (x) assign z = y;
  else assign x = y;
endmodule
</pre>
----
Input:
<pre>
module zy;
if (x) begin:z1
assign x=y;
end
else
if (y) begin:z2
assign z=y;
end
endmodule
</pre>
Output:
<pre>
module zy;
  if (x) begin : z1
    assign x = y;
  end else if (y) begin : z2
    assign z = y;
  end
endmodule
</pre>
----
Input:
<pre>
module zz;
if (x) begin:z1
assign x=y;
end
if (y) begin:z2
assign z=y;
end
endmodule
</pre>
Output:
<pre>
module zz;
  if (x) begin : z1
    assign x = y;
  end
  if (y) begin : z2
    assign z = y;
  end
endmodule
</pre>

fangism, 2020-02-04:
What you've shown looks right.  (and it looks like these are in the integration tests in formatter_test.cc)

closed 2020-02-04 by fangism