chipsalliance/verible #160: alternate style support for formatter [atkenny15]

fangism, 2020-01-30:
Invitation for @atkenny15 to express style differences from current Verible implementation.

Reference: https://github.com/MikePopoloski/slang/issues/291

Useful information:
* An existing style guide would be great
* Examples of desirable formatting

muneebullashariff, 2023-09-27:
Verible also does linting for SV code so I am wondering as how this project is different?

fangism, 2023-10-03:
Not sure if I'm understanding the question, but linting and formatting are different operations.  Linting is read-only analysis, while formatting is transformative.  They are no doubt related when it comes to rules of style.

muneebullashariff, 2023-10-04:
What I mean is, this project **verible** does linting and formatting, so why do we need the new **project slang** used for?

fangism, 2023-11-23:
The slang project includes full semantic analysis, type checking, elaboration, but the Verible project still lacks those.