chipsalliance/verible #161: Incorrect line-break when formatting long clocking block

mvanrompuy, 2020-01-30:
Running verible_format on:
```
  clocking cb @(posedge clk);
    input req;
    input rwn;
    input addr;
    input datain;
    input ack;
    input dataout;
    input error;
  endclocking
```
outputs this
```
  clocking cb @(posedge clk); input req; input rwn; input addr; input datain; input ack; input dataout
      ; input error;
  endclocking
```
I assume the Google style guide states that this should be a one-liner?

Three questions/problems:
- Can I can tune this behavior (as this block has a clear 'endblocking' statement I don't see the benefit of having it all on one line)? Or disable it?
- When this is processed by verible_lint, it complains about the max line length being exceeded (103 characters line length).
- Even when following the one-liner approach, I would expect this as output:
```
  clocking cb @(posedge clk); input req; input rwn; input addr; input datain; input ack;
      input dataout; input error;
  endclocking
```

fangism, 2020-01-30:
This is formatter work that needs to be continued after https://github.com/google/verible/issues/31 .
(Not an issue with our style, just unfinished business.)

Task:
* partition signal declarations to one per-line (indented +2 spaces)

fangism, 2020-01-30:
If you need a temporary workaround to preserve your code (while this gets fixed):

```systemverilog
// verilog_format: off // because of bug#N
... stuff left unformatted ...
// verilog_format: on
```

fangism, 2020-01-30:
If you're interested in fixing this, here's how:

Examine the syntax tree of the input (or reduced version of it), using `verilog_syntax --printtree`.

Abbreviated, you'll see some tree paths like:

```
Node (tag: kDescriptionList) {
  Node (tag: kClockingDeclaration) {
    (#"clocking" @2-10: "clocking")
    (#SymbolIdentifier @11-13: "cb")
    Node (tag: kEventControl) {
      ... (omitted for brevity) ...
    }
    (#';' @28-29: ";")
    Node (tag: kClockingItemList) {
      Node (tag: kClockingItem) {
```

These tags are syntax tree node enumerations from:
https://github.com/google/verible/blob/610fbea3bf873b4ef03b3331827bb21a99b2bc46/verilog/CST/verilog_nonterminals.h#L31

These same enumerations need to be handled in this function:
https://github.com/google/verible/blob/610fbea3bf873b4ef03b3331827bb21a99b2bc46/verilog/formatting/tree_unwrapper.cc#L519

You'll notice that `kClockingItem` and `kClockingItemList` are absent.
See how other items and lists are handled and add these cases.

Then add tests to tree_unwrapper_test.cc that verify the new partition structure on your test case(s).

Then add end-to-end tests to formatter_test.cc.

fangism, 2020-01-30:
b/144605109 (internal reference number)

fangism, 2020-02-02:
@mvanrompuy would you be interested (and have time) in fixing this with the above suggestions?  or rather leave it to one of the (few) developers?

mvanrompuy, 2020-02-03:
@fangism Thanks for the pointers. I'll take a look today and see if I can implement a fix.

commit 651eebfe (2020-02-04): PR #171: Implemented clocking block with ports formatting
Fixes #161

GitHub PR https://github.com/google/verible/pull/171

Copybara import of the project:

  - 335616bce2acabf7cd0f142f20ef9e89f0f8dfc4 Implemented clocking block with ports formatting by Maarten Van Rompuy <maartenvanrompuy@gmail.com>
  - eb41c7d96098e680b129d28275ce4372209132d8 Restored consecutive clocking block case by mvanrompuy <maartenvanrompuy@gmail.com>
  - 449c2a6ea5fed686955650b441c4b150784480a1 Added separate case for clocking block with ports by mvanrompuy <maartenvanrompuy@gmail.com>

Closes #171

PiperOrigin-RevId [... cut]
closed 2020-02-04 by hzeller