chipsalliance/verible #162: Forbid parameter names starting with 'disable', suggest 'enable' instead.

fangism, 2020-01-30:
Our style guide now explicitly discourages naming parameters starting with 'disable'.
The guidance is to use positive sense naming, so starting with 'enable' is recommended.

Task:
* Check all parameter/localparameter declarations for this (even where those keywords are not required, e.g. module parameters inside `#()`).
* Make sure to cover case variations like `DiSaBlE`
* Cover class parameters, package parameters.
* Unit-tests for all of the above.

fangism, 2020-01-30:
b/138755708

fangism, 2020-01-31:
For the section reference (part of diagnostic issued), use the anchor: `#binary-parameters`

tgorochowik, 2020-02-17:
@fangism should this be part of `ParameterNameStyleRule`?

I think it makes sense to create a separate rule for this (so one can be used without the other), but I guess it would require us to rename the currently existing rules to include what they actually check? `ParameterNameStyleRule` is quite broad.

Or maybe the prefix is not really a part of `style` and it could be just called `ParameterNamePrefixRule`?

fangism, 2020-02-17:
Until we have more configurable rules (depending on #133 ), let's keep this as a separate rule to make it easy to disable.  We could call this one something like "PositiveMeaningParameterNames".

commit 30f1c907 (2020-02-21): PR #204: Forbid parameter names starting with 'disable', suggest 'enable' instead.
Fixes #162

This is affected by #203 so I think it makes sense to wait for that to be fixed first.

GitHub PR https://github.com/google/verible/pull/204

Copybara import of the project:

  - 8da43b623fb79013a36a7085f2b9c96ae4ed5bcd Add positive meaning parameter name linter rule by Tomasz Gorochowik <tgorochowik@antmicro.com>
  - 616a4185ca9b041723dbc6292b0b3ad7d8cf1ce0 Add unit tests for positive meaning paramater name rule by Tomasz Gorochowik <tgorochowik@antmicro.com>
  - 4b45b130ee9ce7969465bd135e45ec49441d [... cut]
closed 2020-02-21 by hzeller