chipsalliance/verible #167: verible_lint: Don't require storage type for string parameters

imphil, 2020-02-01:
We explicitly don't specify the storage type `string` for string parameters, as many tools don't support that (these tools are typically fine with string parameters without explicit type annotation).

For example here: https://github.com/lowRISC/ibex/blob/b22fca7bbedb5daa808f475e435632b15c82e4b4/rtl/ibex_core.sv#L22

When we run that through verible_lint, we get:

```
ibex_core.sv:21:28: Explicitly define a storage type for every parameter and localparam, (MultiplierImplementation). [Style: constants] [explicit-parameter-storage-type]
```

As soon as waivers are there (#105) we can waive these errors. But even nicer would be (an option) to disable the type checking for string parameters.

snsokolov, 2020-02-01:
This issue was brought up during the style guide discussion too.

Just for the information:
a) which tools don't support 'parameter string', but support omitted 'string' in your report?
b) do they still correctly synthesize/simulate the code with string parameters used with omitted 'string' in it (like this one https://github.com/lowRISC/ibex/blob/3fcede8a13cb6679866ebccc9a9267235cf2efd4/rtl/ibex_ex_block.sv#L115)?

fangism, 2020-02-01:
I'm aware of at least one report of a tool that doesn't accept `parameter string foo = "...";`.

Until we accommodate variations and special cases, you can waive in-line with:

```
(offending code to left) // verilog_lint: waive rule-name
```

or 

```
// verilog_lint: waive-start rule-name
...
// verilog_lint: waive-stop rule-name
```

I'll add this to the public documentation.

fangism, 2020-02-01:
or to turn off a rule entirely, invoke `verilog_lint --rules=-rulename`

imphil, 2020-02-01:
Cadence Genus doesn't support it at least ([bug report](https://github.com/lowRISC/ibex/issues/429)), and I'm pretty sure I've seen problems with explicitly typed string parameters in other tools as well. (But I don't have hard data at this point unfortunately).

Would it be possible to make the "explicitly type string parameters" a separate rule to disable it independently of the other type checks (which are valuable)?

(On waivers: we cannot use inline waivers because we support too many tools, we need external waiver files.)

snsokolov, 2020-02-02:
> Cadence Genus doesn't support

As far as I remember, when we discussed this we concluded that
a) if the tool correctly understands 'string params', it must support providing the explicit type as well.
b) If the tool doesn't understand 'string params', then omitting 'string' would be a very dangerous workaround. Instead the problematic code should be `ifdef-ed out or removed from the file list.

Would it be reasonable alternative to just `ifdef this code out or remove from the file list?

imphil, 2020-02-02:
We are trying to avoid the ifdef here and have not yet seen negative effects of removing the string type annotation.

Why is leaving it off dangerous?

fangism, 2020-02-02:
b/139804046 (closed, but has some notes on this subject)

fangism, 2020-02-02:
Some users reported that a Vivado tool had the exact same issue, and omitting the `string` type was the supported way moving forward.  They opted to waive this rule to move forward.

I had made a half-joking suggestion to try something like:

```systemverilog
`define STRING    // blank
module foo #(parameter `STRING bar = "baz");
endmodule
```
```

fangism, 2020-02-02:
Once we have some configurability of lint rules from https://github.com/google/verible/issues/128, it would be easier to make an exception for implicitly typed string parameters.

snsokolov, 2020-02-02:
> Why is leaving it off dangerous?

If (b) the tool doesn't support string parameters - this could lead to unexpected results and so is dangerous.

For example, are you giving the following code to the tool? Are you sure the tool correctly handling this if statement?

`if (MultiplierImplementation == "slow") begin : gen_multdiv_slow`

https://github.com/lowRISC/ibex/blob/3fcede8a13cb6679866ebccc9a9267235cf2efd4/rtl/ibex_ex_block.sv#L115

imphil, 2020-02-02:
I never expect a tool to do anything correctly, after all, this is the EDA tool world. While I agree that omitting the `string` type adds risk here, it's not a risk that we are not prepared to deal with. I feel (and I'm not alone with that feeling) that picking a common subset of language features that tools are known to support, and having an ifdef-free code base as result, is a beneficial trade-off.

snsokolov, 2020-02-04:
Just for the record. There is also an option of explicitly typing these parameters as packed arrays, but this unfortunately:
1. requires constantly tracking the length of the string
2. Requires a good linter tool to catch unexpected truncation due to length mismatch, because simulators are not flagging it
So the explicit type is not really helping in this case...

```
parameter bit[7:0][5:0] foobar = "foobar";  // using 5:0 or more will still work
module test;
  initial $display(foobar == "foobar");  // returns 1 as expected
endmodule
```

As for the lint rule exception, we need to think about how to handle cases like `parameter foo = bar;   // where 'bar' could be a  'string' or not.`?

fangism, 2020-03-04:
This will be do-able after we introduce rule configurability in #133 .

fangism, 2020-05-19:
fixed by 9cf082583293deab7b32bca3d244e908717d30d4

closed 2020-05-19 by fangism