chipsalliance/verible #169: Tool names

imphil, 2020-02-01:
*Please see this issue as a very optional suggestion.*

Currently, the Verible-generated tools are called `verilog_lint`, `verilog_format`, and `verilog_syntax`. These are very generic names and don't reflect the fact that the project is actually called Verible. 

While the project is still fresh, would it be possible to rename the tools to something like `verible_lint` and `verible_format`? (Or `verible-lint` and `verible-format` [1])?

---

[1] Rough data point on `-` vs `_` in binary names:
```
philipp@host /usr/bin 
$ ls | grep "_" | wc -l
267
philipp@host /usr/bin 
$ ls | grep "-" | wc -l
756
```

fangism, 2020-02-02:
"Verible" was an internal project name for a few years prior to the open-source release, and it was never certain how acceptable/durable/infringement-free it would be (I was prepared to change it if knowledgeable people advised it), but it passed the (low) bar, and maybe it will stick around.

Today, it is exclusively a Verilog language support project, however, if asked to help support other languages with similar tools, I would re-use almost everything in the language-agnostic `common/` libraries, upon which the Verilog tools are built.  Having this clean separation of language-agnostic/language-specific code made it easier to test a lot of the code without depending on a fully-fledged front-end.   Maybe someday this code bases will mature to the point where separation makes sense, but we're nowhere near today.

Another thing to consider is that the tools support directives like `// verilog_lint: waive ...` and `// verilog_format: off`.  To maintain consistency with a new tool name, we'd have to rename those too.
(The documentation about those features will be published soon to README.md.)

All that said, I can't stop you from renaming/symlinking/aliasing the binaries when you install them.  :)

imphil, 2020-02-02:
I fully understand the hesitation.

Just as one quick note: symlinking is a local solution, but not a scalable one. We will have other scripts and tools calling verible tools, and these will rely on a consistent binary name.

hzeller, 2020-02-21:
I agree that 'verible' should be part of the name. 'verilog_syntax' is super-generic.

How about we prefix the tool names with verible ? Also, I strongly prefer dashes as names.

To have a pleasent experience on my workstation, I'd call them

`verible-verilog-lint`
`verible-verilog-format`
`verible-verilog-syntax`

This makes sure that tab-completion with `verible-`<TAB> works as expected. It contains the relevant information about the tool (verilog and what for) and it does away with the underscores.

If it requires some renaming, so be it, I am happy to do the work to change things. We're still in the early phases when it is simple to change stuff around without upsetting too many people.

fangism, 2020-02-21:
Note to self: any helper scripts that are written/distributed along with verible that call these binaries will need to undergo same transformations.  Something a post-install sed/rename action could take care of.

imphil, 2020-05-28:
@hzeller is this something you're still interested in? The more time passes, the harder such a rename becomes, if desired. Is there something I can help with?

hzeller, 2020-06-04:
@imphil Yes, still interested. Just working out how we can have an intermediate stage where the new names are installed and the old names would still work (either with a little adapter shell-script or something the installer could do)

closed 2020-06-09 by hzeller