chipsalliance/verible #170: Nested structs and typedefs

imphil, 2020-02-01:
Look at this code:

```verilog
  typedef struct packed {
    struct packed {
      logic        q;
    } classa;
  } alert_handler_reg2hw_intr_state_reg_t;
```

https://github.com/lowRISC/opentitan/blob/ebcabe267a093b2eba868f4ca9128826fefff74b/hw/top_earlgrey/ip/alert_handler/rtl/autogen/alert_handler_reg_pkg.sv#L26-L39

Note that adding a nested `typedef` to the second struct is not valid Verilog.

Running verilog_lint on this code gives us:

```
alert_handler_reg_pkg.sv:27:5: struct definitions always should be named using typedef. [Style: typedef-structs-unions] [typedef-structs-unions]
```

Following this advice leads to invalid code (the parser knows that and will error appropriately if there's a nested typedef).

fangism, 2020-02-02:
b/147759348

fangism, 2020-02-02:
Same as https://github.com/google/verible/issues/128
(closing as dupe, continue discussion there)

closed 2020-02-02 by fangism