chipsalliance/verible #172: verilog_lint crash due to "type_id::create($sformatf("instr_%0d", i));"

weicaiyang, 2020-02-04:
I found the style check crashed due to "type_id::create($sformatf("instr_%0d", i));"

Here is [link](https://github.com/google/riscv-dv/blob/f7e35d7939a27ae17b0481eb070e9a36ea335d1f/src/riscv_instr_stream.sv) to the file
```
  virtual function void create_instr_instance();
    riscv_instr instr;
    for(int i = 0; i < instr_cnt; i++) begin
      instr = riscv_instr::type_id::create($sformatf("instr_%0d", i));
      instr_list.push_back(instr);
    end
  endfunction
```

Error message:
> verilog_lint: ./common/util/casts.h:30: To verible::down_cast(From*) [with To = const verible::SyntaxTreeLeaf*; From = verible::Symbol]: Assertion `f == nullptr || dynamic_cast<To>(f) != nullptr' failed.
> xargs: /tools/verible/verilog_lint: terminated by signal 6

weicaiyang, 2020-02-04:
found another case with same failure
```
  virtual function void add_field(string fld_name, int unsigned bit_width,
                                  reg_field_access_t access_type,
                                  bit [XLEN-1:0] reset_val = '0);
    riscv_reg_field new_fld;
    new_fld = riscv_reg_field::type_id::create(fld_name);
    new_fld.bit_width = bit_width;
    new_fld.access_type = access_type;
    new_fld.reset_val = reset_val;
    fld.push_front(new_fld);
  endfunction
```

fangism, 2020-02-04:
I tried to reproduce just now with:

```
$ verilog_lint ~/testing/verilog/issue-172.sv
```

but it doesn't crash for me, I just get:

```
testing/verilog/issue-172.sv:1:11: syntax error, rejected "function" (go/verible#syntax-error).
testing/verilog/issue-172.sv:3:5: syntax error, rejected "for" (go/verible#syntax-error).
testing/verilog/issue-172.sv:3:22: syntax error, rejected "<" (go/verible#syntax-error).
testing/verilog/issue-172.sv:3:36: syntax error, rejected "++" (go/verible#syntax-error).
testing/verilog/issue-172.sv:5:27: syntax error, rejected "(" (go/verible#syntax-error).
testing/verilog/issue-172.sv:6:5: syntax error, rejected "end" (go/verible#syntax-error).
```

Do you happen to have a stacktrace? or any (non-sensitive) tool version information?

fangism, 2020-02-04:
I only tested the snippet from the first comment though.

fangism, 2020-02-04:
Sorry, I retested on the entire file:

https://raw.githubusercontent.com/google/riscv-dv/f7e35d7939a27ae17b0481eb070e9a36ea335d1f/src/riscv_instr_stream.sv

and it doesn't crash for me, just returns success.

fangism, 2020-02-04:
I'm running: `Built on Feb 3 2020 04:57:29` (abbreviated info from `--version`)

weicaiyang, 2020-02-04:
what is the option to show the version. `--version` seems not working
> /tools/verible/verilog_lint --version
> verilog_lint
> Debug build (NDEBUG not #defined)

fangism, 2020-02-04:
Looks like the public bazel build is missing build-stamping.  Can you find out if your verible build is snapshotted to a particular commit-id?

fangism, 2020-02-04:
https://github.com/google/verible/issues/173 is for tracking the version info matter.

weicaiyang, 2020-02-04:
I tried to use latest commit ID to re-install it. This is the script I ran (I changed commit ID)
https://github.com/google/riscv-dv/blob/09dfa95eb81f01fc709271e54def566d320556b8/verilog_style/build-verible.sh

But I got below error. Could you please let me know how to upgrade to the latest version?
> HEAD is now at 03c7102 Some useful public documentation about tool features and usage.
> INFO: Analyzed 1048 targets (11 packages loaded, 2773 targets configured).
> INFO: Found 1048 targets...
> ERROR: /home/weicai/riscv-dv/build/verible/common/parser/BUILD:26:1: undeclared inclusion(s) in rule '//common/parser:bison_parser_common':

hzeller, 2020-02-04:
This might be due to a compiler change; I had the same issue when the system compiler was changed from gcc-8 to gcc-9 on my workstation (you might be in the same situation) but bazel somehow still had cached old header file dependencies.

Since even `bazel clean` didn't do anything, I just checked it out afresh and it compiled.

weicaiyang, 2020-02-04:
@hzeller I remove the bazel and re-install it again. Now I got another error. Any clue? 

> INFO: Analyzed 1048 targets (0 packages loaded, 0 targets configured).
> INFO: Found 1048 targets...
> ERROR: /home/weicai/.cache/bazel/_bazel_weicai/a1f9ac607dc8e1920633b0712a97dd1e/external/com_google_absl/absl/base/BUILD.bazel:163:1: undeclared inclusion(s) in rule '@com_google_absl//absl/base:base':
> this rule is missing dependency declarations for the following files included by 'external/com_google_absl/absl/base/internal/cycleclock.cc':
>   '/usr/lib/gcc/x86_64-linux-gnu/9/include/stdint.h'
>   '/usr/lib/gcc/x86_64-linux-gnu/9/include/stddef.h'
>   '/usr/lib/gcc/x86_64-linux-gnu/9/include-fixed/limits.h'
>   '/usr/lib/gcc/x86_64-linux-gnu/9/include-fixed/syslimits.h'
>

hzeller, 2020-02-04:
I'd also force a fresh cache-building for bazel:
```
mv ~/.cache/bazel ~/.cache/bazel-old
```

weicaiyang, 2020-02-04:
Thanks @hzeller It works. I can re-build verible now.

@fangism I think I have re-build with latest Verible. But I still see below error.
> /tools/verible/verilog_lint src/riscv_instr_stream.sv 
> verilog_lint: ./common/util/casts.h:30: To verible::down_cast(From*) [with To = const verible::SyntaxTreeLeaf*; From = verible::Symbol]: Assertion `f == nullptr || dynamic_cast<To>(f) != nullptr' failed.
> Aborted

fangism, 2020-02-04:
Yes, I suspect this is an actual functional issue, not due to any memory corruption or compiler toolchain.
Would you care to attempt to reduce a test case for us?

weicaiyang, 2020-02-04:
> Yes, I suspect this is an actual functional issue, not due to any memory corruption or compiler toolchain.
> Would you care to attempt to reduce a test case for us?

either cases below can reproduce the error for me.
```
class crash1;
  virtual function void create_instr_instance();
    riscv_instr instr;
    for(int i = 0; i < 10; i++) begin
      instr = riscv_instr::type_id::create($sformatf("instr_%0d", i));
    end
  endfunction
endclass
```

```
class crash2;
  virtual function void add_field(string fld_name);
    riscv_reg_field new_fld;
    new_fld = riscv_reg_field::type_id::create(fld_name);
  endfunction
endclass
```

fangism, 2020-02-04:
Thanks, I'm still unable to reproduce this, but I bet this will reduce further:

```systemverilog
class crash2;
  virtual function void asdf();
    foo::type_id::create(name);
  endfunction
endclass
```

Confirm?

I bet there's an incorrect assumption about the first function argument to `::create` always being a string-literal (or something of that nature).

weicaiyang, 2020-02-04:
> Thanks, I'm still unable to reproduce this, but I bet this will reduce further:
>
> ```systemverilog
> class crash2;
>   virtual function void asdf();
>     foo::type_id::create(name);
>   endfunction
> endclass [...]

above can pass, but I can produce the crash with below code (only add `f = `)
```
class crash2;
  virtual function void asdf();
    f = foo::type_id::create(name);
  endfunction
endclass
```

fangism, 2020-02-04:
Got it.  The fault is certainly in here:

https://github.com/google/verible/blob/master/verilog/analysis/checkers/create_object_name_match_rule.cc

Some invalid assumptions being made about syntax tree structure.

weicaiyang, 2020-02-04:
Awesome. Thanks

tgorochowik, 2020-03-17:
This is the same as #222 and should be fixed by #232

commit 0de6db5f (2020-03-19): PR #232: Fix dynamic_cast failure on [create-object-name-match] rule
Closes #222
Closes #172

The comment in the file actually says:

```
// Returns token information for a single string literal expression, or nullptr
// if the expression is not a string literal.
// `expr_node` should be a SyntaxTreeNode tagged as an expression.
```

So this extra check in fact verifies if the given token can even be a string literal.

The problem in #222 was that the argument was in fact an expression (called as `l("name")` instead of just `"name"`) and the present checks failed to fully verify that.

GitHub  [... cut]
closed 2020-03-19 by hzeller